kylecorry31/Trail-Sense #1516: Feature Request: Backtrack to log location to local file

18fadly-anthony, 2022-10-25:
Backtrack to log location to local file live (not just on export). I use [syncthing](https://syncthing.net/) on my phone so if I could backtrack to a local file that could be synced to my devices live and be used to track my phone's location. Also if [show bluetooth devices](https://github.com/kylecorry31/Trail-Sense/issues/1223) gets implemented it would be cool for that to log to local files as well.

kylecorry31, 2022-10-25:
@18fadly-anthony thank you for the suggestion - unfortunately for Trail Sense to be able to do that, I believe it would need the write external files permission, which I do not want to have requested at this time.

I'll do more research to see if there's a way around it though - maybe have the user pick the file first, though there may be a time limit on how long Trail Sense can write to that for.

kylecorry31, 2023-09-04:
Replaced by #1785

closed 2023-09-04 by kylecorry31