kylecorry31/Trail-Sense #1529: Sensitive notifications

intr-cx, 2022-11-08:
Android notifications are by default private (sensitive), and with a setting these notifications will have their content hidden on the lock screen. 
It'd be nice to have certain notifications from Trail Sense, such as Weather, Today's forecast, Astronomy alerts and Sunset alert, to be set as public so that their contents can be seen at a glance.
Perhaps it could be added as a toggle in settings?

Ps. Thanks for the app. Since I'm often without internet, it's my go-to for all sorts of tools and info.

kylecorry31, 2022-11-08:
Thanks! I'll take a look at what options Android makes available to developers around notification visibility, and I can add a setting in the Privacy settings of Trail Sense to toggle this on and off.

I'll probably default the toggle to public since I don't believe there are any sensitive notifications in Trail Sense (I'll confirm when I pick up this ticket)

kylecorry31, 2026-02-22:
It looks like Android provides the ability to change lockscreen visibility of notification contents via the app's notification settings. I don't plan on changing anything in Trail Sense to add custom support for this.