kylecorry31/Trail-Sense #1535: Notification text has changed from easy-reading, legible English to unreadable, arty nonsense.

cor-of-org, 2022-11-14:
I used to get, "No Change", now I get "unchanging".

I used to see alerts about "rain", now I get "precipitation soon".

Whoever is fucking with the source code, just stop it.

ffs, I don't have time for your childish antics.

kylecorry31, 2022-11-14:
@cor-of-org I can change Unchanging back to Not Changing, I agree that is a bit clearer.

Trail Sense never had an alert for "rain", it used to say worsening (if it did it was in the beta versions a couple years ago) . As of right now, I do not know how to predict what type of precipitation will happen based on pressure and cloud readings (I'm still conducting research into this - if it is even possible to do with limited sensor readings). For this change, I felt that precipitation was more descriptive than worsening since it is commonly associated with a weather front (and was what I was trying to imply with "worsening"). Until I have a research driven method to determine the type of precipitation, I will not display rain/snow.

Also, I am authoring these changes based on research I've conducted into weather forecasting (so there's nobody messing with the source code). I'm always open to constructive feedback though, and can make changes if needed.

If you would rather see "worsening"/"improving" rather than the type of weather (i.e. if precipitation is not accurate in your area), I can add a setting to revert back to that - let me know.

cor-of-org, 2022-11-15:
I just noticed the latest commit.

We know it's a "weather" notification. We don't need to be told again within the notification.

And "not changing" is 20% good, whereas "no change" is rock-and-roll, 100% w00t!w00t! on the nail.

Hire a poet or something to sort out your text.

kylecorry31, 2022-11-15:
@cor-of-org thank you for providing some context - I didn't realize that unchanging would be more difficult to read than no change (to me it appeared fine, but I also have the context of being the one who made the change in the first place). Unfortunately, given that I work on Trail Sense for free, I do not have the resources to hire someone to help out with text/UX - if something is unclear, I rely on people like you to let me know (thanks!). I would definitely be open to hearing your ideas on how I can improve the weather icons (I've gotten so used to them that they are easy to understand for me, but there's also developer bias in that).

My latest commit did not add the word "weather" to the notification - the weather_not_changing is just the resource identifier for translation purposes. It currently says "Not changing soon", I can change that to "No change soon", thanks.

As for precipitation - I wish it were that easy :)  The thermometer within a phone is impacted by the battery and CPU temperature, therefore it is very unlikely it will every record a temperature near freezing (there may be ways to better calibrate it, I just need to learn more about it). My idea was to use historical temperatures for your location/time of year to do a similar method (also not a trivial task). I want to make sure the route I take for making that prediction is backed by solid research.

kylecorry31, 2022-11-16:
Thanks!

I would love to just query a weather data provider for temperature, but I took the stance early on with Trail Sense that it would not have the ability to access the Internet. This is something I can't take back without losing the trust of many users, and it also has the benefit of allowing the app to work in areas with spotty or no Internet connection (I made this app for my backpacking trips in areas of no cell signal). The downside is that I have to get a bit creative about how Trail Sense uses on-device sensors, which is why I can't trust the thermometer.

kylecorry31, 2022-11-16:
I am going to close this ticket - if you feel I missed something, please let me know and I can re-open it or create a new one.

The change I made was converting "Unchanging" to "No change", and then I've created two more issues:
- Revisit weather icons (https://github.com/kylecorry31/Trail-Sense/issues/1538)
- Determine type of precipitation (https://github.com/kylecorry31/Trail-Sense/issues/1539) - dependent on an improved temperature estimate

closed 2022-11-16 by kylecorry31

cor-of-org, 2022-11-16:
Ouch! Okay, gotcha. You got some tricky science ahead!

In fact, I'd say there isn't the computing power, currently, in even the best Android device, to measure all the factors which could influence your results. As they say, good luck!

But this is maybe something that the user could meantime have some say in..

If I'm out back-packing or mountain-biking and want accurate information, it's up to _me_ to get the conditions right for that; i.e. if I want accurate temperature-related information, _I_ get the device free of obstructions (or thermally-influencing factors - Me, I keep my phone in an identical side pocket on (standard webbing) my backpack (standard webbing); whichever pack I'm wearing - not in a thermally-lined pocket, I mean) - so ambient temperature readings should be good. But it would _also_ be nice to be able to choose internet providers for local temperature-related information; at least until this becomes (as I get farther into the wild) unavailable. Besides; what's happening in my near vicinity isn't usually the same as what's happening up in the atmosphere, where weather is made.

I think it's great that you are building something that works properly offline. But at the same time, I want a tool that uses *all* sources of information, and if some information becomes available, even for a minute, I'd like an app that takes this into account.

Clearly you are someone who likes a challenge!

kylecorry31, 2022-11-16:
Yes, lots of tricky science ahead - but I'm definitely enjoying the challenge.

I've been exploring the idea of Trail Sense companion apps for some time. That is, add-ons to Trail Sense which can be downloaded separately. So for the privacy focused users, they never have to worry about Trail Sense using the Internet, while others could download something like a "Weather Provider Expansion" which polls a weather service to get some data that Trail Sense can process and cache. I think this would be the best of both worlds - Trail Sense can then use the most accurate data it has access to.

But with my limited time to work on Trail Sense, the companion apps have been pushed aside for other core features. Before I forget, I created this ticket to track a weather provider expansion: https://github.com/kylecorry31/Trail-Sense/issues/1540

cor-of-org, 2022-11-18:
Today I got, "Unchanging soon". lol

kylecorry31, 2022-11-18:
Do you think that will read better as "no change soon"? I can see for the "later" cases it would be clearer as "no change until later", but I'm not very clear on what to put for "no change soon".

kylecorry31, 2022-11-18:
Actually, thinking more about it, I'm going to remove that text for the "no change" case - I don't think it makes sense to show it when no change is detected, because unlike the other weather conditions, it does not really have a good indication of how long there will be no change.