kylecorry31/Trail-Sense #1540: Weather provider expansion

kylecorry31, 2022-11-16:
A companion app which polls a weather provider at a set frequency and provides that information to Trail Sense for further processing and caching. 

This would be a standalone, optional app download that requires Trail Sense be installed.

michaelblyons, 2023-08-07:
Sync a common packing list with file share? 😅

kylecorry31, 2023-08-07:
@michaelblyons can you clarify what you mean? 

My interpretation of you comment is syncing to a service like Dropbox/NextCloud for others to use? Or do you mean just the ability to import/export a packing list (in which case, that is something I could implement without a plugin)

michaelblyons, 2023-08-07:
That's pretty close. I think I saw that you're planning ex/import to CSV for packing lists. The file could be synced with Bluetooth, yes.

I am coordinating with my partner: "Did we already pack enough X?" It is mostly preparing the vehicle for 1-day, 3-day, or 7-day trip that would be good to share with a friend. When we get there, every adult manages their own list for excursions. If TS is not the right tool for shared packing, we can surely find something else.

If people use packing lists to track stocks on distance hikes as well, having a merge algorithm for intelligent incr/decr might be helpful. But I am not really doing that in the backcountry.

kylecorry31, 2023-08-07:
Thanks! I'll consider adding that along with the offline file share plugin

MikaelaLovett, 2024-03-26:
That walkie-talkie feature would be so useful for me 🙈 Sorry for "spamming" with comments, but I found your app yesterday and I'm very excited about it, haha

I'm disabled and we use Zello with my dad (when I need help getting out of the shower, can't get up, my heart rate is too high etc.) and he can't hear me through the closed door. But Zello is WiFi only, so if the internet goes out (happens frequently) he needs to pick up my call first without hearing my message immediately and he can miss it if his phone's on mute (media sound is always on) ✨

kylecorry31, 2024-03-26:
> That walkie-talkie feature would be so useful for me 🙈 Sorry for "spamming" with comments, but I found your app yesterday and I'm very excited about it, haha
> 
> I'm disabled and we use Zello with my dad (when I need help getting out of the shower, can't get up, my heart rate is too high etc.) and he can't hear me through the closed door. But Zello is WiFi only, so if the internet goes out (happens frequently) he needs to pick up my call first without hearing my message immediately and he can miss it if his phone's on mute (media sound is always on) ✨ Edit: I also struggle to talk loudly or at all in certain situations due to trauma, so that'd be great during medical emergencies or even when we're out and about geocaching 🗺️

I'll have to take a look at Zello - my hope is that the walkie talkie will support both wifi (direct connection) and bluetooth (maybe even at the same time, using whichever has the best signal). It could even have the option of sending a message and having it read out load on the receiving device (using text to speech).

MikaelaLovett, 2024-03-26:
> I'll have to take a look at Zello - my hope is that the walkie talkie will support both wifi (direct connection) and bluetooth (maybe even at the same time, using whichever has the best signal). It [...]

Oh my, thank you for considering it, the text to speech feature would be life changing! 💗

kylecorry31, 2024-05-13:
Initial experimentation with the comms plugin can be found here (extremely work in progress - basically an experimentation project at this point): https://github.com/kylecorry31/Trail-Sense-Comms

MikaelaLovett, 2024-05-13:
> Initial experimentation with the comms plugin can be found here (extremely work in progress - basically an experimentation project at this point): https://github.com/kylecorry31/Trail-Sense-Comms
> 
> Still have not determined what the plugin architecture will look like / what is allowed on Google Play Store

Thank you for working on it! 🍀

kylecorry31, 2024-05-18:
Possible architecture:

[image]

Under this model, the tool would live in Trail Sense and the comms plugin just adds the ability to talk with nearby devices.

Example flow of the walkie talkie:
1. User installs comms plugin, "Local Talk" tool shows up in TS
2. User opens the tool and is prompted to connect to a device
3. User clicks connect button which launches the comms plugin activity to select a device
4. The comms plugin requests bluetooth permission
5. User selects the device they want to connect to and a foreground service is started
6. User is returned to Trail Sense where they interact with the tool - TS uses AIDL to communicate with the comms plugin
7. User stops communication -> TS uses AIDL to stop the comms plugin
8. User returns to app and selects an existing connection -> TS starts comms activity with pre-selected device -> process repeats

This will also allow other tools to leverage the plugin, such as the ability to see nearby devices on the navigation/photo map tools or to send notes/files.

leavenode, 2024-07-24:
METAR/TAF functionality would be useful as far as weather plugins go. Perhaps the ability to translate raw data as well? How about a database of all known airports? A METAR/TAF widget perhaps?

kylecorry31, 2024-07-24:
> METAR/TAF functionality would be useful as far as weather plugins go. Perhaps the ability to translate raw data as well? How about a database of all known airports? A METAR/TAF widget perhaps?

Thank you for the suggestion, I will consider that as part of the Provider plugin

kylecorry31, 2025-06-14:
I don't yet have a release APK, but https://github.com/kylecorry31/Trail-Sense-DEM-Plugin is the start of the DEM plugin. Once downloaded Trail Sense will detect it and unlock the DEM and DEM + Barometer altimeter modes.

15 arc-second resolution, with about 30 meter vertical resolution for the first version (100 MB). I have tiles for 15 arc-second, full vertical resolution but thats almost 700 MB, so I would need to have a way to download those tiles over the reduced resolution version. Not sure if GitHub is the right place to host those, I'm not looking to charge for this so I don't want to pay hosting costs for a server.