kylecorry31/Trail-Sense #1605: request feature: backup and restore app data for non-root users

Ananas-Kuchen, 2022-12-12:
Hi, I accidentally uninstalled the app without realizing any app data and my custom for settings belong to root path. any data about pedometers, notes, qr codes i saved were completely erased

is it possible to add backup and restore option for all this app data?

first is the option to backup/restore original app data from root which saves settings and more in a zip

second is option to export, any notes/qr code notes and others to text format (.txt). however, those text files can also be imported into this application again

kylecorry31, 2022-12-12:
Hi @Ananas-Kuchen ,

I'm sorry the uninstall also removed your settings. I have an existing issue for the full export of TS, which I plan to do sometime this year (probably between the major features on my roadmap). I'm going to close this issue out in favor of that one: https://github.com/kylecorry31/Trail-Sense/issues/743

closed 2022-12-12 by kylecorry31

Ananas-Kuchen, 2023-01-06:
@kylecorry31 ok. too bad

I checked with emulator on windows, app data of this app is in databases folder for all recorded data and custom settings are in shared_prefs folder. I sometimes find developers who make an export app data feature without converting it to another file, it's really the same as from the root folder. but there are also those who convert it to another format. it's just an example

hopefully sometime in the future you can make it happen if possible. now at least the feeling of being stuck unable to move app data will still occur

thanks for the response. and sorry because I made a duplicate issue 🙏

kylecorry31, 2023-01-07:
Yeah, due to the way Trail Sense is set up, I plan on exporting to another file - I plan to implement this after my work on the Offline Maps tool is complete.