kylecorry31/Trail-Sense #1612: Feature request: Custom morse code

kylecorry31, 2022-12-17:
This was suggested by a user on Google Play.

Allow flashlight and whistle to play a custom morse code message.

More research is needed before I decide to add this feature:

- [ ] Do search and rescue teams use morse code to communicate with the person they are trying to rescue? If so, why and how often?
- [ ] What benefits would this add to Trail Sense? (ex. long distance communication at night with a flashlight - but is that really necessary + users would need to know how to read morse code)
- [ ] Are there any other use cases which this would be beneficial for? (adjacent to TS' use cases)

Portagoras, 2023-02-19:
I think it would be interesting for long distance communication, especially if you would have a way to decode messages as well. Maybe by using the camera with a "scan" field, similar to the QR-Code readers, where an algorithmy is decoding the brightest flashes as morse code. This would make communication over a distance at night possible, even without any network. Would probably be a different issue, but just something that came to mind.

kylecorry31, 2023-02-19:
@Portagoras interesting idea, I will need to experiment with how accurately the camera can pick up flashes from a phone's light at night over long distances

michaelblyons, 2023-12-01:
> ...using the camera with a "scan" field, similar to the QR-Code readers, where an algorithm is decoding the brightest flashes as morse code. This would make communication over a distance at night possible, even without any network.

I was also wondering this.

The app linked in #804 does not appear to ingest light from the camera. I have *no* idea whether it can even be done reliably.

anon2828226, 2024-01-24:
Flashdim has a custom morse code flashlight output feature if you want an example of an implementation.
https://github.com/cyb3rko/flashdim#features

detecting morse code with a camera seems like a harder problem.

ManDay, 2024-11-21:
I think an automated morse flash like flashdim's is a nice gimmick and together with an automated receiver it would make for a fancy tool. That said, I don't think it belongs in Trail-Sense lest the latter become a "and-the-kitchen-sink" kind of toolbox.

Bundling a set of a "survival utitlities" in a single app makes sense to me. But at some point we have to admit that there is no good reason to add something to that toolbox, if it can rightfully be a separate and independent app. I think this applies here, because, ultimately, there are only two versions of this:

- A complete visual morse "radio" app with emitter and receiver, which just seems to go way beyond the scope of Trail-Sense and should rather be a dedicated app
- Only the emitter, which, practically speaking, wouldn't be more than gimmick, because (from my impression) it would be used by people who know morse in the first place, in which case the textual input with complete control over Q-codes, speed, and timing is more effort than just hitting the button yourself (therefore #2696).

Just my opinion. It would make a fancy tool, but I think flashdim would be a better place for it.

kylecorry31, 2024-11-21:
@ManDay thank you for your opinion, I'll factor that in when I review this issue. I'll also take a look at flashdm as a recommended app.

Since this was suggested by multiple users I want do thoroughly investigate it's practicality and use cases. I had similar thoughts with it belonging in Trail Sense (I'm of the opinion that only SOS is useful for the majority of situations that Trail Sense supports).

kylecorry31, 2024-12-27:
After further consideration, I will not be adding custom morse code to Trail Sense. The majority of people do not know morse code other than SOS (and even that is not well known, just stands out). I've added FlashDim to the recommended apps for users who want to send morse code messages.

closed 2024-12-27 by kylecorry31