kylecorry31/Trail-Sense #1618: Premium version on Google Play (donation only)

kylecorry31, 2022-12-26:
Multiple users have asked that I create a premium version of Trail Sense - no added features, just a way to donate via Google Play.

Not sure if I'll do this - I feel like some users may mistakenly think it will have more features, but I'm adding this issue to check if other developers are doing a similar thing and if they have any bad reviews about it being donation only (and to check Google Play's policies around this).

Essentially, it's just a convenient way for people to make a donation from their phone if they wanted to.

JensKorte, 2023-02-11:
* <https://f-droid.org/de/packages/net.osmand.plus/>

kylecorry31, 2023-02-11:
Thanks! I don't plan on removing the free version from Google Play if I do this, but those might be helpful to see how I can setup different tracks.

kylecorry31, 2023-09-04:
I don't think this will be worth the confusion that there would be nothing different between the versions, there's nothing I would want to put behind a paywall, and I don't want to deal with additional Google Play policies around selling apps at this point.

I'm switching away from donations in general and more towards just asking for feedback and new ideas to make things easier.

closed 2023-09-04 by kylecorry31