kylecorry31/Trail-Sense #1639: Manual Temperature Input

RageGamerBoi, 2023-01-17:
My phone (like many others I assume) has a temperature sensor only for its internals.

To get a temperature reading more accurate than the historical temperatures, a manual input to extrapolate the daily highs and lows from is probably best. It should also probably gradually converge with the historical temperature over long periods of time with no input.

Even if one doesn't have a thermometer on them, most people can estimate temperatures pretty accurately, so this would be a good feature to have.

kylecorry31, 2023-01-17:
Thanks for the idea! I originally planned to allow a manual override in the historic temperature update, but I scrapped it because I couldn't think of a good way to allow it to be varied.

I like the concept of gradually reducing it back to the historic temperature, I think that resolves my issues around the manual temperature. 

I'm think this could play into #1570 - instead of a manual thermometer source, it would be a form of historic temperature calibration (I don't see any benefit in applying this to the temperature sensor as well).

RageGamerBoi, 2023-01-17:
Because it would change over the course of only a few days, a button to quickly set it should probably be on the Weather page (only if enabled ofc)

This keeps it from being the set it and forget it thing that thermometer calibration currently is.