kylecorry31/Trail-Sense #1665: Fill status bar with background color

kylecorry31, 2023-02-10:
A user requested this via Google Play - instead of showing a black bar at the top, have the background color under the status icons.

kylecorry31, 2024-02-11:
> Hi @kylecorry31, i would like to help with this improvement.
> 
> I've already started working on it, but I wanted to ask you if you prefer the statusbar to be consistent with the bottom bar or should it only be consistent with the bg? (this is especially noticeable with the theme set to 'dark')
> 
> Here a video example of the result by matching the statusbar to the colour of the bottom bar:
> 
>  Screen.Recording.2024-02-11.at.16.09.18.mp4

Thanks! I would say matching the background color for the statusbar (top) and matching the bottom bar color for the system navigation bar would be preferable. Though if that is difficult to do, then I think what you have is definitely a step in the right direction.

AlessandroFrangiamone, 2024-02-11:
Ok, perfect, I opened the PR https://github.com/kylecorry31/Trail-Sense/pull/2218 with the change as requested.
Attached to the PR is also the video result.

kylecorry31, 2024-02-11:
Thanks, your PR has been merged and this will be available in the February release.

closed 2024-02-11 by kylecorry31