kylecorry31/Trail-Sense #1683: [Research] Solar compass

kylecorry31, 2023-02-24:
Research using the sun as a compass if the magnetometer does not work.

I'll have to do more research on the practicality of this - the astronomy tool can display the direction of the sun or moon (click on the sun/moon icon on the chart), so I imagine this would be pretty similar. If the magnetometer isn't available, I would assume the gyroscope would also be unavailable or unreliable to use, but it's worth a test.

JensKorte, 2023-02-24:
https://github.com/ligi/SurvivalManual/wiki/DirectionFinding (I guess the link is obvious ;) )

leavenode, 2023-03-15:
I see a new option to display the sun and moon on the compass in the most recent update. Good that :-)

Would it make sense to also have an option where you can rotate the compass dial manually? 

My Moto E7 (inexplicably) has no compass sensor, but a couple of my older phones have them but they are twitchy and unreliable and they make the dial jump around. Would be nice to just bypass them completely.

leavenode, 2023-03-15:
Ah, I see the compass smoothing option. Helps with the jitter, but not the general wonkiness.

kylecorry31, 2026-11-02:
This will be too much work for very little benefit. Moving to closed with the potential of a future revisit

closed 2026-11-02 by kylecorry31