kylecorry31/Trail-Sense #1696: Copycat/Clone of Trail Sense on Google Play

Poussinou, 2023-03-14:
Hi @kylecorry31 

I found this app https://play.google.com/store/apps/details?id=com.sladjan.smartcompass on the Google Play Store from [this developer](https://play.google.com/store/apps/dev?id=4678864934774732136). This guy just took your source code and put it on the Store, without warning this is free software...

As you are the owner of the source code, feel free to open a request to google [here](https://support.google.com/legal/contact/lr_dmca?product=googleplay) to remove this copy from the Store, if it's your wish ;)

Don't forget to tell Google that:

* The app is licensed under MIT license and this guy is violating it;
* He is falsely claiming he made the app;
* Provide some screenshots of your own app.

kylecorry31, 2023-03-14:
Thank you for letting me know - I haven't seen that version of Trail Sense in a while (that's several years old).

I'm not going to request it gets taken down without giving the developer a chance to correct the copyright violation (they just need to include Trail Sense's license in the app - I couldn't find it anywhere). I know Google Play has a policy against copycat applications, but since this is such an old version of Trail Sense (with a changed color scheme), I don't think that will be a problem.

kylecorry31, 2023-03-14:
The developer has said they will update it within 48 hours to include the proper licensing, at which point I will resolve this issue.

closed 2023-03-20 by kylecorry31