kylecorry31/Trail-Sense #1704: Barometer calibration

kylecorry31, 2023-03-21:
- [ ] Set offset directly
- [ ] Set offset via current pressure
- [ ] Create CalibratedBarometer wrapper

leavenode, 2023-05-28:
It will be good to get this working. I can calibrate the barometer on my Casio ProTreks(I typically use my local METAR as reference) but the readings on my various phones are still all over the place.

kylecorry31, 2023-05-28:
@leavenode I will plan to get this done after the Photo Maps feature is release in around a month

For the most part, Trail Sense does not need it perfectly calibrated to predict weather - most of the weather prediction works on the change in pressure rather than the absolute pressure. That's why this issue was a lower priority. The only part that uses the absolute pressure is the "later" weather prediction - and that does not require it to be accurate either (within 10-15 hPa of actual should be fine). It's also a similar case with the altimeter (which already supports calibration) - that just uses changes in pressure from the reference pressure/elevation, so an accurate absolute pressure is not a strong requirement.