kylecorry31/Trail-Sense #1715: App recommendation: Briar

kylecorry31, 2023-04-01:
For offline communication

michaelblyons, 2023-04-19:
Bluetooth communication in the woods? Wouldn't you be within earshot for it to work? Maybe helpful for hearing impairment, I guess.

kylecorry31, 2023-04-20:
Hi, the Walkie Talkie idea was a user suggested feature - I haven't done much research into its practicality (ex. range) yet.

https://github.com/kylecorry31/Trail-Sense/issues/1540 documents other users of bluetooth that could be possible, such as offline file sharing or shared navigation.

michaelblyons, 2024-03-26:
@MikaelaLovett You may be interested in this issue. Briar doesn't currently do voice, but they have an [issue discussing it](https://code.briarproject.org/briar/briar/-/issues/985). The nice thing is that they switch between communication modes.

I also found an app called "WalkieDroid," that purports to do Bluetooth push-to-talk, but I have not tried it.

MikaelaLovett, 2024-03-26:
> @MikaelaLovett You may be interested in this issue. Briar doesn't currently do voice, but they have an [issue discussing it](https://code.briarproject.org/briar/briar/-/issues/985). The nice thing [...]

Ohh thank you very much, I'll check it out!! 💗

closed 2024-12-27 by kylecorry31