kylecorry31/Trail-Sense #1719: Dynamic backtrack intervals

soundray, 2023-04-10:
I would like backtrack intervals to depend on recent movement.

Background: I currently have the backtrack interval permanently set to 30 minutes. This way I record a useful long-term location history while not draining the battery too much. However, when I'm cycling or hiking, I still use another app (OsmAnd+) for fine-grained tracking. If I forget to stop this tracking, I get excessive power consumption and huge log files.

I think it would be really cool if I could use Trail Sense for both purposes. I would like it if the tracking interval would stay long when I'm at home or at work (not moving, or moving slowly in inconsistent directions), but would shorten dramatically when I travel. A bonus would be if interval shortening would be triggered by movement as picked up by the acceleration sensors.

kylecorry31, 2023-04-10:
Thank you, this is a good idea. 

Notes to self:
Look into using the significant motion sensor + GPS speed to adjust interval when user starts moving. If not significant motion, than just GPS speed could work (it would have a delayed start though)

kylecorry31, 2024-07-05:
Thank you, I will bump the priority of this a little bit.

For reference, TS does not officially support cycling, skiing or driving related activities, so this is mainly for accounting for breaks during a hike. I don't think significant motion will help too much with cycling, so GPS speed will be more beneficial in your case.

closed 2026-02-22 by kylecorry31