kylecorry31/Trail-Sense #1738: Discrepancy between circular and scalar compasses(?)

hyprdrm, 2023-04-23:
Hi.

First things first: this is a _great_ app, one of the best I've found in a long time. I'm still discovering the features, but I love it already; as a keen walker with a new 'phone and a concern for personal privacy, this ticks a lot of boxes for me :)

So kudos and thanks to the dev.

---

There seems to be a discrepancy between:
* the "rosette" (i.e. circular) compass display (the one that's used while the 'phone is flat)
* the "gauge" (i.e. scalar) compass display (the one that's used while the 'phone is vertical)

The latter correlates closely to my physical (Silva) compass; the former is consistently ~10˚(?) above the latter.

Is that intentional, is it user error, or is it a bug?

(I've tried correlating the compass by doing the traditional "figure of eight" dance; it doesn't make any difference).

Thank you.

---

Google Pixel 6
GrapheneOS (Android 13)
Trail Sense 4.17.0

kylecorry31, 2023-04-23:
Hi, thank you for the feedback!

I can't seem to reproduce this issue - I just tested using a couple real compasses and all of Trail Sense's compass views matched up (Pixel 5A, Android 13, 4.17.0).

Can you elaborate how you are testing the compass accuracy so I can make sure I'm doing it the same way? 

My test methodology is as follows:
1. Place the compass on a piece of paper and mark its north on the paper (easy with a baseplate compass to use its straightedge to trace a line).
2. Replace the compass with Trail Sense and see if it lines up
3. Repeat for different orientations - I find it much easier to do with the round/rosette compass because I can line me phone up with the north line perfectly

If I place the phone and compass side by side, the phone's compass changes to point closer toward the physical compass - removing the physical compass allows the phone to line up to the line I drew for north. There is much less of an effect when the phone is vertical because the sensor is further from the physical compass' magnet and at a diluted angle (in fact I don't see any distortion unless I hold the physical compass near the top of my phone).

Also, be sure that both Trail Sense and your compass are using the same declination value. If your compass isn't configured for declination, you can turn off True North while testing under Trail Sense Settings > Sensors > Compass > True North.

hyprdrm, 2023-05-01:
Thank you for looking into this - I appreciate it - and apologies for a delayed response.

> My test methodology is as follows:

(snip)

That sounds admirably simple and robust; thank you 👍

If I repeat it, though, I see the same behaviour as before: the rosette shows 357˚ (which I thought was close enough) but the scale shows 15˚. The discrepancy seems _slightly_ reduced if I enable "Use legacy compass"...but not significantly: it's still ~15˚.

Am I doing this wrong, maybe? (I disabled "True North" in Settings > Sensors > Compass, and Trail Sense says "Reported accuracy: Good").

Thanks again.

kylecorry31, 2023-05-01:
@hyprdrm So if I understand correctly, after following my test methodology, the rosette compass aligns with North properly, but the linear compass does not? If I understood your initial test, it was the opposite (rosette compass didn't align) - can you confirm which one isn't matching up?

Since there was some introduced error in my methodology in regards to testing the linear compass, I repeated my test but this time I used a straight edge to perfectly make my phone perpendicular to the North line I drew while holding it vertical, and it perfectly matched the value reported by the rosette compass (both at 2 degrees).

I also tested both with declination enabled/disabled (compasses matched at about 347 degrees while pointed at magnetic North), and with the legacy compass enabled/disabled (compasses matched).

Can you try the following:
1. Try to repeat that test, but this time using a straight edge with the linear compass (I've never been a fan of that compass mode due to how hard it is to line up perfectly)
2. Try doing the figure-8 calibration but this time holding your phone vertically (so the linear compass is shown)

It's also possible that this is due to GrapheneOS - unofficial ROMs aren't officially supported by Trail Sense, and it's possible that there's something lacking in the ROM that's causing the calibration to not work correctly (or be reported correctly). I have an old Android phone that has GrapheneOS on it, I'll test that out after.

kylecorry31, 2023-05-01:
A quick preliminary test after updating my GrapheneOS phone (Pixel 3A), shows that there is definitely something weird going on with the calibration there (it also keeps switching between Good and Unknown). The rosette compass looks fine, but the linear compass is off by 10 degrees. I'll do more research this week before I say for sure this is a GrapheneOS issue, but there may not be much I can do to handle it on Trail Sense's end.

hyprdrm, 2023-05-01:
> So if I understand correctly, after following my test methodology, the rosette compass aligns with North properly, but the linear compass does not?

Yes. (Apologies if I was unclear before).

> A quick preliminary test after updating my GrapheneOS phone (Pixel 3A), shows that there is definitely something weird going on with the calibration there

I'm glad it's not just me!

Thanks again for looking into it.

kylecorry31, 2023-05-01:
Yeah, I'm almost certain this is a bug within GrapheneOS. The phone I tested on was the one I've been using to develop Trail Sense until about a year ago, at which point I switched it to GrapheneOS and set it aside. It was definitely working properly under stock Android. I'll want to run a few more tests before I say there's nothing I can do from Trail Sense's end (short of implementing its own magnetometer calibration, which doesn't seem feasible).

Notes to self:

Before closing this feature as a system bug, I want to check the following.

- [ ] Is the issue reproducible using another compass app?
- [ ] Is the issue reproducible when using legacy compass? (since the azimuth is calculated by the system and not TS - if that's wrong, I don't think it's possible for TS to ever be right)
- [ ] Inspect raw sensor values (gravity, magnetometer) to see if there's anything that stands out as incorrect
- [ ] Document issue in FAQ if it is a bug in GrapheneOS (could be a bug in other unofficial ROMs as well)
- [ ] Is there any workaround to improve the calibration on GrapheneOS?

kylecorry31, 2023-09-04:
Closing this as my research showed this was only happening in GrapheneOS and seems to have been improved with the new compass types (sensor fusion with gyro)

closed 2023-09-04 by kylecorry31