kylecorry31/Trail-Sense #1755: Allow more than right angle rotations for maps

kylecorry31, 2023-05-10:
- [x] Enhanced image view allow other angles
- [ ] Auto rotate sets exact angle
- [ ] Allow manual rotation of exact angle
- [ ] Adjust bounds to match angle
- [ ] Enhanced image view allows pan and zoom while rotated

Z3NOX, 2023-05-27:
One question to address the Magnetic North calibration by rotating the map: Do you already somehow have access to  the magnetic declination based on the current position in Trail Sense? Or in other words: How would I retrieve the information on how much I have to rotate my map in order to switch from a Magnetic to a True North orientation?

Another thought which came to my mind: Wouldn't it be possible (and even sometimes easier) to get the orientation angle by another calibration point? In my (maybe naive) thinking the rotation of the map is only one degree of freedom which could be compensated with one additional information. Or am I missing something here?

kylecorry31, 2023-05-27:
@Z3NOX you are correct, in the latest version of Trail Sense (4.18.0) I did just that (except it rounds to the nearest 90 degree angle, which is why I removed the step to manually rotate the map). You only need two calibration points for the rotation to be resolved (comparing pixel angle to real world angle). I was experimenting with this earlier in the month for non-right angle rotations, and did in fact see it compensate for the magnetic north vs True North in my testing (there's still other things that I need to address as part of this issue for it to be ready to release).

And yes, you can find declination values in Settings→Sensors→Compass, though it's a little tricky to see what it is at different locations (I just search for it online - doing it in TS would involve mocking the GPS via the GPS override setting).

closed 2023-06-08 by kylecorry31

kylecorry31, 2023-06-22:
@Z3NOX I wanted to let you know that this was released over the weekend, I did a lot of my testing against the map you provided and it seemed to be working well. Thank you for providing that!

Z3NOX, 2023-06-22:
Yeah I just received the update via F-droid today and fiddled around a bit. Seems to work great! Is there a way of knowing/displaying the rotation angle? I noticed the north symbol in the upper right corner which seems to show the True North (which in itself is a bit odd, as it displays a magnetic compass…) but was not able to get to the precise information.

Glad I could help and if I notice something else I'll let you know.

kylecorry31, 2023-06-22:
There isn't currently a way to show the rotation angle, but I'll will add that: #1822 . I'll also see if I can change that compass icon to something clearer - as that always points to True North (I'll check out what other maps apps are doing, and do the same).

Under Trail Sense Settings > Photo Maps, you can disable "Keep map facing up" so that the top will point to True North rather than keep the map at a right angle. It doesn't change any of the beacons/paths/navigation, but can give a clearer idea on how much the map is rotated. There are some issues with that though (mainly when you pan/zoom it's a little odd - not a breaking issue though #1791 ).