kylecorry31/Trail-Sense #1780: Photo Maps calibration feedback missing in 4.18

Z3NOX, 2023-05-25:
First, thank you for this wonderful app which I really enjoy using! I am using the experimental photo maps feature for some time now and stumbled upon the changes made in [release 4.18](https://github.com/kylecorry31/Trail-Sense/releases/tag/4.18.0).

Generally I have found, for my use cases, that the calibration is only valid locally which leads for me to recalibrating the map regularly. Therefore, I found it useful to have all paths and beacons lying in the map section rerendered during calibration to check the quality of the new calibration. With changes in 4.18 this "feature"(?) seems to be gone which leaves me kind of blind during the calibration process. 

Is there a way to get the path- and beacon previews back in the calibration process? (e.g. via settings)

kylecorry31, 2023-05-25:
@Z3NOX thank you for the feedback - I removed that functionality because it was imposing some technical limitations on the map calibration. 

I will try to add that back in for the full release with some adjustments to make it work better with my plans for calibration.

kylecorry31, 2023-05-25:
Notes to self:

- [ ] Reload beacons / paths when calibration changes
- [ ] Display beacons and paths once map has enough calibration points (with opacity)
- [ ] Extract MapView logic into a base class or property (prefer a property so it isn't tied to a single map image for future tiling support)
- [ ] Extract beacon / path loading
- [ ] Handle when calibration points indicate map rotation changes (don't rotate the map, only the layers)

Z3NOX, 2023-05-25:
Do you need further feedback on the Photo Map feature and its calibration? I could provide you with a fairly detailed map + Backtrack + Beacons + GARMIN-Track. Tracks and Beacons match well but I have had no luck with calibrating the map [globally] correctly. I don't know whether it is a projection problem (63°N) or something else. I would open another issue for this if you want.

kylecorry31, 2023-05-25:
Yes, that would be great - the maps I have calibrated (42°N) have been very accurate, except for one that was designed using magnetic North for some reason (#1367). Feel free to email them to me at trailsense@protonmail.com if you don't want to add them to a GitHub issue (if you do want to put it on GitHub, I would recommend creating a new issue around map calibration accuracy).

commit 90290710 (2023-08-23): Add map layer preview during calibration

Closes #1780
closed 2023-08-23 by kylecorry31