kylecorry31/Trail-Sense #1781: Photo Maps calibration issue

Z3NOX, 2023-05-25:
Using the calibration in the Photo Maps feature I am not quite able to nail the matching on the whole map. It is always just a good local approximation. 

I used a [public orienteering map (PDF)](https://github.com/kylecorry31/Trail-Sense/files/11567659/karta_2_arnas_friskaovik_2023.pdf) ([source](https://7an.se/friska)) and logged the following [GPX files](https://github.com/kylecorry31/Trail-Sense/files/11567703/GPX.zip):

- trail-sense-beacon-arnaes.gpx - Beacons saved to Trail Sense, used in the calibration tool to match the center of the circles in the map
- trail-sense-backtrack-arnaes.gpx - 1-minute-backtrack created with Trail Sense 
- trail-sense-GARMIN-arnaes.gpx - separate GPS log with Garmin Instinct

I saved a beacon for some of the POIs indicated by numbers (23-30) on the map. The tracks and the beacons match well together which I take as an indicator that the GPX accuracy was not too bad (although it was in the woods). I also tried to change the projection in the tool's settings as I could imagine that the difference in 63°N is quite large however, I could not notice a difference there. 

PS: I post processed the GPX-files with [Viking](https://sourceforge.net/projects/viking/) to anonymize the timestamps and hope this does not cause any inconveniences.  If so, just let me know.

kylecorry31, 2023-05-25:
Thank you, I will take a look and let you know what I find

kylecorry31, 2023-05-26:
@Z3NOX It appears this is another instance of the map being generated using magnetic north rather than True North. I'm going to see if it is feasible for me to do #1367 or #1755 before I release the Photo Maps (maybe even delaying the release).

For reference, I used Caltopo with the beacons you provided to generate a geospatial PDF - this worked perfectly in Trail Sense. So I took that map image, and overlayed your map in a photo editor. Notice that your map aligns with the magnetic north arrow that was generated by CalTopo (declination of 8.5 degrees), rather than the True North arrow.

[image]

kylecorry31, 2023-05-26:
For reference, here's the Caltopo PDF: https://caltopo.com/p/5QK10

closed 2023-05-26 by kylecorry31

Z3NOX, 2023-05-27:
Yeah, on a second thought it seems kind of right for an orienteering map to be directed to magnetic north as they want to help your orientation with a compass… And I can confirm that rotation of the map by 8.5° fits the recorded data very well.

I would rename the issue such that it can be recognized to be related to the Magnetic North subject, even if it is closed, maybe it is useful for later reference.