kylecorry31/Trail-Sense #1784: [question] Where is data stored?

MarcelRobitaille, 2023-05-29:
I would like to get my path data out of Trail-Sense to create a dashboard with my recent trips.

I would like to know where the path data is stored. I checked in `Android/data` and expected to see a folder `com.kylecorry.trail_sense` in there, but I did not. Is there any way to get access to the paths via the filesystem?

kylecorry31, 2023-05-29:
Hi, the paths are stored in a database that is private to the Trail Sense application on Android. On my phone (debug build), it can be found under data/data/com.kylecorry.trail_sense/databases/trail_sense. It uses a SQLite DB.

But if you are just looking to export the paths to a GPX file which can be imported into other apps, you can do that directly from the Trail Sense application. On the Paths list (Tools > Paths), click the 3-dot menu on the right of the path you would like to export, and select Export from the menu. This will let you export the path GPX file to a location on your device that you have access to (ex. Downloads).

MarcelRobitaille, 2023-05-29:
I see. I assumed that the same `.gpx` files would be somewhere in the phone's filesystem.

Exporting would work, if there were a way for this to happen automatically. As far as I know, there is no option in the application to automatically export paths to a folder of the user's choice. Is this something you would consider implementing?

Here's some more information on my use-case: I use Trail-Sense to track my jogging (I know it was designed for hiking, but it's the best app I've found on F-Droid for jogging). I want to create a dashboard that tracks which days I go jogging, how long I go, how far I go, my average speed, and my weight. That is why it has to be automatic. If I have to export each path manually, I will never remember. My plan is to sync the paths to my server using nextcloud and have some application on my server to parse these files and generate the dashboard.

kylecorry31, 2023-05-29:
Thanks for the details and the support!

I'll do more research into it to see if it is something I can do.

Exporting multiple paths: https://github.com/kylecorry31/Trail-Sense/issues/1270
Auto export paths (new): https://github.com/kylecorry31/Trail-Sense/issues/1785

Also, while I personally don't use it, I've heard good things about OpenTracks - which is more focused on activities such as jogging and cycling. It also seems to have automatic export set up. https://f-droid.org/packages/de.dennisguse.opentracks/

MarcelRobitaille, 2023-05-29:
Thanks for the recommendation. I'll check that out!

kylecorry31, 2023-06-10:
I'm going to close this issue in favor of the two other issues mentioned in my last comment

closed 2023-06-10 by kylecorry31