kylecorry31/Trail-Sense #1796: Redesign: Navigation

kylecorry31, 2023-06-07:
I think this UI is pretty simple and easy to understand, though some tweaks can be made.

- [ ] Make the navigation button clearer (actually put the word navigation in there)
- [ ] Hide the status icons behind a minimized version if they are all good
- [ ] Improve the sighting compass UI
- [ ] Improve the linear compass (make it feel more responsive)
- [ ] Improve the radar compass (make it less cluttered - ex. I don't think the circles are actually useful)
- [ ] Improve the round compass (make it easier to read at a glance)
- [ ] Improve navigation panel (not exactly sure yet, I'm just not a huge fan of it)

mjray, 2025-05-29:
Can I beg you to be cautious about this? The radar-compass navigation screen is my most-used bit of the app and it works pretty well for me.

If I understand what you mean about the circles, I find them useful for estimating how far away I am from a turn in the path/route that I'm following (if my range is set to 200m, then I think each circle is 50m).

One thing I would find useful is more options for what's at the top of the radar compass. When I'm on my steel bike, the device compass is pretty useless (large lump of magnetic metal too close to it). While GPS compass would help, being able to fix North = Top would useful to stop the display flipping around as the bike turns (or wobbling a lot if the smoother is set low). I think Sun = Top and Moon = Top could also be useful in some situations.

kylecorry31, 2025-05-29:
> Can I beg you to be cautious about this? The radar-compass navigation screen is my most-used bit of the app and it works pretty well for me. Great app overall and the other features are great bonuses.
> 
> If I understand what you mean about the circles, I find them useful for estimating how far away I am from a turn in the path/route that I'm following (if my range is set to 200m, then I think each circle is 50m).
> 
> One thing I would find useful is more options for what's at the top of the radar compass. When I'm on my steel bike, the device compass is pretty useless (large lump of magnetic metal too close to it). While GPS compass would help, being able to fix North = Top would useful to stop the display flipping around as the bike turns (or wobbling a lot if the smoother is set low). I think Sun = Top and Moon = Top could also be useful in some situations.

Thank you for the suggestions, I will be cautious when adjusting this if I ever do a redesign of this tool.

I would like more information about the suggestion around locking things to the top. I'm not sure I'm following what that suggestion is. Maybe a rough sketch/mockup would help?

mjray, 2025-08-04:
Hi. Here's an attempt to make a rough mockup. The idea is that there would be a new button (in the centre bottom here, above the true-north star, and green here but maybe yellow or grey would make more sense, although green/amber/red might indicate the confidence of a GPS-derived direction) that would show what was currently fixed to the top of the display (GPS-derived direction, North, Sun, Moon (not shown), or Compass (not shown)) and the radar changes appropriately. I'm not sure whether the centre arrow would make sense in all of them.

GPS-derived direction at top:
[image]

North at top:
[image]

Sun at top:
[image]

closed 2026-02-19 by kylecorry31