kylecorry31/Trail-Sense #1802: 3D sun/moon position preview

kylecorry31, 2023-06-09:
- [ ] Add a 3D button to the sun/moon chart
- [ ] Tapping 3D button opens a 3D view of the sun / moon position and their paths
- [ ] Add slider at bottom of view to change time
- [ ] Use augmented reality view

This requires #1702 to be completed

michaelblyons, 2023-10-29:
It occurs to that you could put time-stamped markers on the prediction arc. That could help a user picture things like

- When the setting sun will reach the western mountains (ie non-horizen sunset).
- When the moon will rise above the trees.

Obv you could also mark degrees and peak, but the time aspect might be particularly useful.

(Apologies for the suggestion if this is already in the build. I haven't tried the nightly.)

kylecorry31, 2023-10-29:
Good idea, I'll add that - I have a text indicator for the "focused" (centered) AR point, I'll include the time, elevation, and bearing in that and look into maybe displaying a couple of the times directly over the camera.

closed 2024-01-07 by kylecorry31