kylecorry31/Trail-Sense #1805: Release Photo Maps (Full)

kylecorry31, 2023-06-10:
- [ ] Remove experimental flag
- [ ] Remove feedback prompt on open

michaelblyons, 2023-06-22:
It's already great.

kylecorry31, 2023-06-22:
@michaelblyons thanks!

I was worried people would find it a little difficult to use, which is why I left it in beta. So far I haven't had any complaints, so that's a good sign (though I suspect not many have tried it out yet). I also want to have #1780 in place, which I plan to complete for the next release.

maverick74, 2023-06-22:
I've personally been using it for a few releases/months and - i must confess - barely read the instructions...

I think it's pretty easy to use iyam (specially in the latest releases).
It's also a very useful tool!

I've been using it to try to find some missing landmarks, but i don't seem to be able to... still, **i think** it's not a software problem but a GPS reception problem, as there are some trees around that **i think** are influencing the reception...

kylecorry31, 2023-06-22:
@maverick74 thank you, glad to hear!

It could very well be a GPS reception problem, or it could be that they are plotted on the wrong place on the map (I've seen some local maps that had things plotted wrong).

Something you could try just in case is to recalibrate your map on the latest release (5.0.0) - you don't need to adjust the points, just click Done on the calibration (With the map open, click the top menu > Calibrate, then just click Done without any changes). Release 5.0.0 brings support for automatic rotation correction, which can handle maps that aren't exactly pointing to True North (I've encountered a couple like that). Not sure if that's the case here, but it may help (it will display a message if the rotation was corrected).

maverick74, 2023-06-22:
> Something you could try just in case is to recalibrate your map on the latest release (5.0.0) - you don't need to adjust the points, just click Done on the calibration

Thanks!! :) Will try next time i get to the site :)

michaelblyons, 2023-06-22:
One thing I've been doing that didn't seem obvious at first is calibrating from points on a map app on the phone, rather than grid lines or coordinates marked on the photo map itself.

- Plunk down a pin 📍 in Google Maps (or OsmAnd or whatever) at an obvious marked location like 🅿️ or a visitor center.
- Copy the coordinates to clipboard.
- Save them in TS to a beacon.
- Mark the beacon on the photo map.
- Repeat for another location on both maps.
- Now it's calibrated.
- (Then you can add other beacons for other places only on the photo map.)
- Depending on the source data, you may want to recalibrate from phone GPS when you get there.

I think the photo map calibration instructions could use some other tweaking, too, but like the poster above, I don't think I used them much when I first tried it.

michaelblyons, 2023-06-23:
The Photo Map feature allows you to use a photo as a map in Trail Sense. Always carry a physical map for backup and verify the accuracy of Trail Sense.

This tool is currently in beta, so please report any issues or feedback at trailsense@protonmail.com.

## How to create a photo map?

1. On the Tools tab, click Photo Maps
2. Tap the plus button on the bottom right and choose how you want to import your map.
    1. **Camera** - Take a picture of your map, making sure to include the entire map in the picture
    2. **File** - Select a file from your device (JPG, PNG, or PDF)
    3. **Blank** - Create a blank map given a location and the size of the map. This is useful if you have created paths or beacons.
3. Crop the photo to the map bounds.
4. Calibrate the map using 2 known locations:
    1. Enter the real-world location in the input field. For example, a trail sign, fork, or point of interest.
    2. Tap the map where the location is.
    3. Use Previous/Next to switch between calibration points.
    4. You can click Next + Done to save the calibration at any time.
5. Tap Done to save the calibration and use the map.

**Note:** If you choose to use a blank map or Geospatial PDF, your map will automatically be calibrated. A good source for Geospatial PDFs is [CalTopo](https://caltopo.com).

## Tips for calibration points 

### Calibrating while hiking
- Point 1: Tap the 🛰️ at the trail sign where you took the photo of the map or the trailhead.
- Point 2: Tap the 🛰️ at the first trail fork you encounter. If you don't encounter a trail fork, use a point of interest on the map (ex. a lake, peak, or other landmark).

### Calibrating from beacons
If you already have marked beacons in the map area, tap the 📍, select the beacon, and drag it to the location on the map.

### Calibrating from the map grid
If your map has gridlines, you can follow the 'Using printed maps' guide to get the coordinates of several features on the map and enter them into Trail Sense.

The intersection points between gridlines are the easiest to calibrate from.

### Calibrating from an online source
If you have a map app with coverage for the Photo Map area (either cached or online), you can look up the coordinates of a map feature and paste them into the Trail Sense calibration box.

michaelblyons, 2023-06-23:
Tweaked markdown above. Particular note: `a.` isn't technically markdown, so all the second level ordered lists were in a single line in 5.0.0

closed 2023-08-20 by kylecorry31