kylecorry31/Trail-Sense #1833: Weather: correct pressure using GPS height

Pandapip1, 2023-06-29:
Pretty simple. I work in a pretty tall building. Every morning, I get a storm alert due to the 5 hPa pressure drop, which is annoying since it also messes with detection in general. I suggest that the weather detection use sea level pressure.

kylecorry31, 2023-06-29:
Hello, can you provide the following information:

1. What version of Trail Sense are you using? Trail Sense Settings > Version
2. Provide me the following values from Trail Sense Settings > Sensors > Barometer
  a. Is "Sea level pressure" enabled
  b. What value are you using for "Smoothing"

By default Trail Sense converts pressures to sea level using GPS elevation. This is meant for outdoor use only, since GPS is not going to get a good reading (if any) from indoors. Unfortunately, if your settings look good, there's not much else I can do since indoor usage is not supported by Trail Sense - both from a technical (GPS does not work well in buildings) and use case (Trail Sense is only designed to support hiking / outdoor use). My recommendation is to disable the weather monitor while in that building, and re-enable it when you go on a hike, etc.

Pandapip1, 2023-06-30:
1. The latest F-Droid version: 5.0.0
2. Here you go:
  a. Sea level pressure: Enabled
  b. Smoothing: 0%

kylecorry31, 2023-06-30:
Thank you for the details.

- Unfortunately, cell tower location will not help with elevation detection, that only provides a rough location in terms of latitude and longitude.
  - See #432 and #972 about network location
  - See #1540 about using a digital elevation model rather than GPS elevation
- As for the compass, yes please log an issue for that with details to help me reproduce the issue. The compass accuracy is directly reported by Android, so there might not be much I can do unfortunately.
  - Before logging a new issue, see #159 which may be related if you would prefer to add details there.
- I don't plan on adding inside building detection since that is beyond the supported use cases of Trail Sense - but I believe the adjustments I make as part of this issue will help. Supported use cases can be found [here](https://github.com/kylecorry31/Trail-Sense/wiki/Use-Cases).

For reference, while not officially supported I've found that GPS elevation does work well enough inside a house or small building (especially when near the windows). This is sufficient for Trail Sense's official use cases, which may include entering a small hut or a cabin while backpacking. The cause of your issue is because when Trail Sense can't get a new GPS reading, it falls back to the last known location/elevation. In most cases this is fine, since it is rare to happen outdoors and the elevation likely won't change much. In your case, the building is large and likely does not get a GPS signal (and if it did the elevation would be way off). To Trail Sense, the elevation has not changed and but the pressure did which indicates a low pressure system moving in.

michaelblyons, 2023-07-08:
I also get a daily storm notification on my commute, although mine is a lat/long change, not straight up in an elevator.

- Sea level enabled
- Smoothing 15%

If I should change some setting to filter the storm alerts, please let me know.

kylecorry31, 2023-07-08:
@michaelblyons I believe this is a similar scenario, just a different way to get there. GPS elevation accuracy decreases with speed and proximity to buildings (depending on where your commute takes you), therefore I imagine the elevations are inaccurate while converting to sea level, and if they are consistently inaccurate it may interfere with the smoothing. Also, depending on how far your commute is, the sea level pressure may actually be different - in that case, Trail Sense isn't able to distinguish that from a storm since it looks identical. In the future #1427 and #1491 may help.

Other than the filtering of inaccurate readings, this scenario falls outside of the use cases of Trail Sense, so I won't likely add special handling for it.

You may see some success modifying the sensitivity of the storm alert to Low (Settings > Weather > Storm sensitivity), which requires a more rapid change in pressure to trigger the notification.

If you want, I can take a look at the data to confirm this is a case of inaccurate elevations. Within 48 hours of this occurring, you can generate a weather export under Settings > Weather > Export weather readings. This produces a CSV file with times, pressure, elevation, elevation accuracy, temperature, and humidity - it does not include GPS coordinates. If you would like to provide that, feel free to upload it here or send it to me at trailsense@protonmail.com

kylecorry31, 2023-07-13:
@michaelblyons after some more thought (and forgetting to turn off backtrack myself), I'm going to make a slight change which may help ignore the invalid readings while driving. https://github.com/kylecorry31/Trail-Sense/issues/1858  and https://github.com/kylecorry31/Trail-Sense/issues/1859

The rationale for this within Trail Sense's use cases is that you may record a hike and forget to turn off backtrack (or weather monitor). As for the weather monitor, a valid use case would be that you use it at two different hikes that requires travel in between - in that case, you wouldn't want to assume that the weather is the same at each location and therefore the weather forecast should restart at the new location (I may experiment with a distance / time based threshold since if the places are close in distance/time, they may still have the same weather).

I believe that will resolve your issue of the storm alerts appearing while driving - I don't expect it to impact the original issue here about it not working properly in buildings, though it could help a little.