kylecorry31/Trail-Sense #1864: Clarify Temperature Calibration with new Source information

mr-ransel, 2023-07-15:
The new temperature calibration source field and "update recorded temperatures" button could use some clarification either in the User Guide, or on the calibration page. 

Specifically, I'm not clear on what changing the Source does exactly or why I'd choose one over the other. I gather it is doing... something with switching how temperature readings are estimated between the battery/ambient sensor and the location-aware historic values but I'm not totally clear on how or what it does or if manually measuring and setting calibration values myself overrides the surmised conclusions drawn based on the Source field.

I'm sure it's a great functionality, it could just use a little clarity in the docs so users can understand and decide which setting is a better fit for their usage pattern.

kylecorry31, 2023-07-15:
Thank you for the suggestion, I'll update the docs. In the meantime, here's what I plan to put (it will be on the thermometer calibration user guide, which I can link on that settings page). Let me know if this addresses what you are looking for.

## Source
On most phones, it is recommend to use the historic temperature source as this will generally be more accurate than the built-in sensor.

All temperatures sources can be calibrated to adjust the temperature if it is inaccurate.

- **Historic**: Use the 30 year historical high / low temperatures (Climate Normals) to estimate the current temperature for your location. Most of the time this should be close to the actual temperature, but if there's a heatwave or storm, it may be inaccurate. Calibration can be used if the temperatures seem very inaccurate for your location, though in most cases it shouldn't be needed. You can preview the underlying historical temperatures by using the Climate tool in Trail Sense.

- **Sensor**: Use the phone's built in thermometer to read the current temperature. On most phones, this sensor will read much hotter than the outside temperature as it is impacted by your phone's CPU and battery. It is recommended calibrate this sensor for everyday use. If you skip calibration, you will get a more accurate reading if you don't use your phone for a while and keep it out of your pocket.

## Updating recorded temperatures
If you change the temperature source to "Historic", you will have the option to update recorded temperatures. This will replace all temperatures recorded by the weather monitor with the estimated temperature. You can use this if you previously recorded temperatures with the sensor, and switched to historic temperatures.

kylecorry31, 2023-07-19:
In addition to the documentation changes I will also add:
- [ ] A link to the user guide from the thermometer calibration fragment
- [ ] A button to reset the thermometer calibration (matches updated documentation)

commit 7194780d (2023-07-19): Add user guide to thermometer calibration page

Closes #1864
closed 2023-07-19 by kylecorry31