kylecorry31/Trail-Sense #1902: Create a better metric for hike difficulty

kylecorry31, 2023-08-08:
The current metric factors in speed and overall elevation gain, but a hike can also be difficult if it contains steep sections.

Idea 1:
- Calculate the Shenandoah National Park metric (easy, moderate, hard) as it is currently doing
- If the trail contains sections that are very steep, increase the rating to moderate or hard (depending on how steep they are / how often they occur)

michaelblyons, 2023-08-13:
I once took a baby (strapped on) and two young children (walking) on a hike that SNP rates "Easy." It was not a great experience for them.

Based on that, I would encourage you to show the "points" of the hike next to the label, and click to see the chart for context.

kylecorry31, 2023-08-13:
@michaelblyons yeah, I'm surprised on how popular that rating metric is now that I'm getting some feedback on it. Most of the hikes I do are either actually easy (local to my town) or hard (mountains) - and I would agree with the SNP ratings on all of them.

Can you clarify what you would like me to show for the points?

For reference, if you switch to the slope point style (right most dropdown above the path map - defaults to None), it color codes the points based on slope. The elevation chart will also color code by slope (always enabled - assuming there's elevation data on the path).

michaelblyons, 2023-08-13:
I just mean the numeric score before it's translated into labels.

kylecorry31, 2023-08-13:
Thanks - the SNP score does have a numeric rating that I could display (maybe upon clicking the rating), but it will get tricky after I apply the second step that factors in slope during the path (unless I use an off the shelf rating system)

kylecorry31, 2023-08-29:
My initial take at a custom algorithm will be in tonight's nightly build

kylecorry31, 2023-09-01:
The algorithm I've landed on is based on the rating system described here: https://www.smcgov.org/parks/county-hiking-trail-difficulty-rating-system

It tries to quantify what that system describes.

Easy:
- Less than 2 miles long
- Maximum slope is under 15 percent
- Gain and loss are under 250ft

Moderate:
- Between 2 and 4 miles long
- Maximum slope is between 15 and 25 percent
- Gain or loss (whichever is greater) is between 250ft and 750ft

Hard:
- Over 4 miles long
- Maximum slope is over 25 percent
- Gain or loss (whichever is greater) is over  750ft

It will check for hard conditions first, then moderate, and finally easy.

This should rate trails harder if they have a steep section even if the whole trail is pretty flat - which was the biggest complaint of the SNP rating.

closed 2023-09-01 by kylecorry31