kylecorry31/Trail-Sense #1911: New Feature Ideas

kylecorry31, 2023-08-11:
This is a continuation of #59, since that was getting a

Please feel free to post new feature ideas in the comments on this issue. I would love to have some other thoughts about what should be included in this app (no guarantee that I will include them, but I will consider each idea). No idea is a bad idea here, worse case I may say that I will not include it in Trail Sense (but who knows, if it is a really good idea, I or someone else may work with you to build it into a separate app). I should also mention, that I am not averse to the idea of having separate companion apps to Trail Sense with specific niches if a good feature does not meet the goal of Trail Sense. 

Some set in stone goals of this app:
- Trail Sense must not use the Internet in any way, as I want the entire app usable when there is no Internet connection
- New features must provide some benefits to people using the app while hiking, in a survival situation, etc. 
- New features should make use of the sensors on a phone rather than relying on stored information such as guides
- New features must be based on peer-reviewed science

For reference, some things I have considered in the past and removed were:
- "Survival blueprints" (recipes for making items while in survival situations)
    - There are a lot of other resources out there and it is best presented in a book format
- Survival guide
    - There are a lot of other resources out there, including this [awesome app](https://f-droid.org/en/packages/org.ligi.survivalmanual/)
- Animal behavior
    - I can't say there is enough science / supporting evidence behind Solunar theory for me to make accurate predictions of animal behavior in general, though if someone provides strong evidence I will consider it

vkuehn, 2023-08-16:
@kylecorry31 the fusion of these two awesome apps would be great

kylecorry31, 2023-08-16:
@vkuehn is that in reference to the heart rate tool and Trail Sense?

If so, I don't plan on merging the two (see the reasoning here: https://github.com/kylecorry31/Trail-Sense/issues/59#issuecomment-1672355144), but I could maybe use this as one of the plugin experiments (shows up in Trail Sense if the second app is installed)

kylecorry31, 2023-09-06:
@Z3NOX thanks - I provided some details in #1908 regarding this feature.

I've also fixed the first post here - no idea where the rest of that sentence went :)

lucahttp, 2023-09-28:
Hi, would be nice to have a Garmin IMG reader to use as Google maps offline,
there is already a lot of work in each country developing this kind of maps for old GPS devices

kylecorry31, 2023-09-28:
@lucahttp thank you, I've updated #604 to include that format.

diggernet, 2023-10-08:
Hi Kyle, thanks for Trail Sense, it's a great app!

I was going to suggest export/import of settings.  It wasn't that big a deal until I started creating Packing Lists and realized that the data is trapped.  But then you beat me to it, adding backup/restore.  Thanks!  Would still be nice to have a way to share individual lists, but that's rather less critical.

I played with Tides a bit, but found it rather tedious to get set up.  It would be handy to be able to enter a station ID and have that station set up or updated from NOAA Tides & Currents.  But of course that violates the "No Internet" rule (which I value).  Then I saw the tide provider plugin idea in #1540.  Beat me again.  Only thing to add is that such a plugin could operate in a couple different modes:  Use the plugin to preconfigure select stations for later use offline, or use the plugin online to find and display the tide for the nearest station.

But I think you haven't beat me to all of these, for Photo Maps:
- Keep PDF maps in a vector format for better zooming.  Or at least (perhaps optionally) rasterize at a higher resolution.  #1345 would certainly help here.
- You mention that Photo Maps supports Geospatial PDFs, such as those from CalTopo.  I have tried using Geospatial PDFs from [USGS Map Locator](https://store.usgs.gov/map-locator), such as [Block Island](https://prd-tnm.s3.amazonaws.com/StagedProducts/Maps/USTopo/PDF/RI/RI_Block_Island_20210309_TM_geo.pdf), but for some reason they are not recognized.
- Ok, here's the fun one:  Those USGS PDFs contain layers (Optional Content Groups, in PDF lingo), not all of which are turned on by default.  It would be great to be able to easily turn different layers on and off while viewing the map.  Among the layers are a shaded relief layer and an orthoimage (offline satellite view!) layer.

kylecorry31, 2023-10-08:
@diggernet thank you for the suggestions!

I agree that the Tides tool isn't ideal without being able to load from the Internet. In addition to #1540 I also have https://github.com/kylecorry31/Trail-Sense/issues/1102 and https://github.com/kylecorry31/Trail-Sense/issues/1104 which should help with this.

Here's new issues to track the map features:
- https://github.com/kylecorry31/Trail-Sense/issues/1988
- https://github.com/kylecorry31/Trail-Sense/issues/1989
- https://github.com/kylecorry31/Trail-Sense/issues/1990

kylecorry31, 2023-10-22:
I'm looking for help around improving the messaging on devices without a compass. If anyone wants to help with some ideas, I would really appreciate if you add a comment to this issue: https://github.com/kylecorry31/Trail-Sense/issues/2010

JensKorte, 2023-10-29:
Provide an emergency function
(This is how I think of it, not the technical details of Android) Android has an emergency button and apps can provide a hook to it and get started in an emergency/panic situation.
TS could provide a hook and start the whistle, the screen and the flashlight sending SOS. For saving battery there could be something like "stop after e.g. 15 minutes and give signals again at every full hour or every full and half hour for five minutes". 

F-Droid supports such a function in the setting. Maybe you can reuse source from there?!

This also touches https://github.com/kylecorry31/Trail-Sense/issues/1471

kylecorry31, 2023-10-29:
@JensKorte thank you, I've added this issue to track this feature request: https://github.com/kylecorry31/Trail-Sense/issues/2027

ojppe, 2023-11-17:
A good feature would be a [sound level meter](https://en.wikipedia.org/wiki/Sound_level_meter)
Something like the metal detector layout looks perfect for this.
Check out OpenNoise on the [Play Store](https://play.google.com/store/apps/details?id=it.piemonte.arpa.openoise) and on [Github](https://github.com/Bustami/openoise-meter-android)

kylecorry31, 2023-11-17:
@ojppe thank you for the suggestion. Can you suggest some use cases for a sound level meter that fit within [Trail Sense's use cases](https://github.com/kylecorry31/Trail-Sense/wiki/Use-Cases)?

I believe someone had suggested that in the past, and at the time I couldn't find any use cases (I can't seem to find the conversation around that - might have been through email or is lost in the old "New Feature Ideas" issue).

ojppe, 2023-11-17:
Determine the loudness of things in the environment (thunderstorm, volcano, waterfall, generator, aircraft, train, music, gunfire) [to protect hearing](https://en.wikipedia.org/wiki/Health_effects_from_noise)

kylecorry31, 2023-11-17:
@ojppe while I think that would be a useful app, I don't think that will fit within Trail Sense's use cases

ojppe, 2023-11-18:
I don't see why not, Trail Sense has become my digital swiss army knife and a sound and speed meter are the only things missing (I can think of) that use most phones' sensors. But you know best!

kylecorry31, 2023-11-18:
@ojppe I apologize, I should have stated my reasons more clearly. I know a lot of the tools in Trail Sense can be used outside of hiking, but I'm trying to limit the scope of what I include in the app to keep it pretty focused. There's general criteria for new features in the first comment of this issue and the use cases wiki, but here's the specifics for a decibel meter:

1. I don't believe hearing protection falls into the supported use cases of Trail Sense: hiking, camping, backpacking, and geocaching. I believe a decibel meter would fit much better into a health, concert, or safety regulation app. The cases where this might come into play while hiking are very rare, and most people who find themselves in this situation will likely not have hearing protection with them.
2. I don't think this tool would get much use in the hiking space, but it would be time consuming to develop and support.
3. The microphone permission is not something I want to add at this point unless there's a feature which will provide a clear benefit in the hiking space for the majority of users.

As for the speed meter, the speed is shown on the Navigation tool in Trail Sense (GPS by default), and also available on the Pedometer tool (based on steps).

If there's something that can be predicted / identified using sound that is beneficial while hiking, I could add this in with that (more of an indicator that it is picking up audio), but as of right now I don't know of anything that falls into this case other than #1143

ojppe, 2023-11-18:
Thank you for the clarification

Wumbo-11, 2023-12-12:
I have an idea for a "_pairing_" feature on this app, it may sound unappealing but it's something you might want to consider for later.

I'm thinking of this feature as being able to allow Trail Sense app users (2 or more people) to pair their coordinates or maybe a '_special location code_' via gps that can be monitored offline. (You know, maybe it's like a bluetooth pairing between connected cell phones and being able to share each device's camera).

It's just that this uses a gps code base that is specifically used in the use of this application only, maybe it could be by scanning a special barcode or giving a special mark in this application. (hope you can understand what I mean)

Well, your app is very useful so far, I will always wait for future updates.

kylecorry31, 2023-12-12:
> I have an idea for a "_pairing_" feature on this app, it may sound unappealing but it's something you might want to consider for later.
>
> I'm thinking of this feature as being able to allow Trail [...]

Thank you for the suggestion. I do have a similar feature planned as part of the Trail Sense Comms plugin - no timeline yet for that though. I hope to do some experimentation on the plugins in 2024, because I think they would be pretty cool and open a lot of possibilities. https://github.com/kylecorry31/Trail-Sense/issues/1540

Saijin-Naib, 2023-12-15:
I would love to see other tools have Android Quick Action tiles alongside Pedometer and TrackBack, like Flashlight, screen flashlight, whistle, etc.

kylecorry31, 2023-12-15:
> I would love to see other tools have Android Quick Action tiles alongside Pedometer and TrackBack, like Flashlight, screen flashlight, whistle, etc.

Thank you for the suggestion, I will get more added. Here's an issue to track them: https://github.com/kylecorry31/Trail-Sense/issues/2120

LeftyDextrous, 2024-02-01:
Hi Kyle,

It would be nice to have an automatic "night-light" timer, where the app theme changes from the user-selected default, which used during the daytime, to the Night theme after sundown. 

This would be useful in situations where you don't want your phone to automatically brighten when you check it during the night, both to preserve your night vision (in an astronomy context) and to avoid spooking animals if you're hunting. Hopefully this should be easy to implement given that Trail Sense automatically determines sundown times anyway.

Thanks for the app,
Lefty

kylecorry31, 2024-02-01:
> Hi Kyle,
> 
> It would be nice to have an automatic "night-light" timer, where the app theme changes from the user-selected default, which used during the daytime, to the Night theme after sundown.
> 
> This would be useful in situations where you don't want your phone to automatically brighten when you check it during the night, both to preserve your night vision (in an astronomy context) and to avoid spooking animals if you're hunting. Hopefully this should be easy to implement given that Trail Sense automatically determines sundown times anyway.
> 
> Thanks for the app, Lefty
> 
> EDIT: I realize now there is a Sunrise/Sundown function. However, I still would like to be able to customize the specific theme to be used during the day and night. Apologies if this FR is redundant.

Thank you for the suggestion! I created the following issue to track that feature: https://github.com/kylecorry31/Trail-Sense/issues/2192   (it is not redundant and would be cool is Trail Sense let you choose which theme to use during the day and night)

kylecorry31, 2024-02-17:
> This surprisingly doesn't exist in any OSS app but I think it would fit in nicely with the existing featureset:
> 
> An alarm that goes off when my location approaches a destination.
> 
> Example 1: You sleep on the bus, and it goes off when you're close to your destination.
> 
> Example 2: Alarm goes off when you get home/work reminding you to do something
> 
> Maybe it would make a bit more sense as part of a map app, but naturally when hiking this would be useful

Thank you for the suggestion, that feature is planned as part of https://github.com/kylecorry31/Trail-Sense/issues/1908 (or https://github.com/kylecorry31/Trail-Sense/issues/2159)

LeftyDextrous, 2024-03-06:
I don't believe this has been requested, so I'll put it out there: can you add a feature that allows the user to hide tools they don't need? It would be helpful to reduce clutter in the tools menu as I have dedicated apps for many of the offered features, for instance the unit converter QR code scanner, and the battery, and the astronomy and navigation tools can be accessed directly rather than through the tools menu. Of course, this is not to say that these features should be hidden by default since they aid in feature discovery, but rather the user should have the option to either hide them entirely or at least banish them to the bottom of the tools screen so they don't clutter the rest of the interface.

kylecorry31, 2024-03-06:
> I don't believe this has been requested, so I'll put it out there: can you add a feature that allows the user to hide tools they don't need? It would be helpful to reduce clutter in the tools menu [...]

Thank you for the suggestion, I will consider adding it here: https://github.com/kylecorry31/Trail-Sense/issues/2255

My thought process on the current tools tab was that the user can pin the tools they use often rather than disable tools they don't use. I figured that there would be about 4-5 commonly used tools, and the rest would either only be used in specific cases (infrequent) or never used (there as a backup in case needed). I'll do more research into allowing tools to be hidden and the implications of that (#2046 is why navigation, astronomy, weather were added to the tools tab).

MikaelaLovett, 2024-03-26:
Hi there, as I wrote in my Google Play review, I really like this app, thank you so much for working on it! ✨ I was wondering if adding local photos (and perhaps notes or files) to the packing list would be possible in the future, please? I struggle with memory loss so every packing is quite stressful. I currently use Bring me for the packing purpose as it has photos (they help me remember which item I meant to bring with me/I own), but with this feature I could use only yours app (I love the weight option!) by itself. Bring me doesn't feel as intuitive and even tho it's obviously a great app, I prefer Trail Sense and it's other features.

Thank you once again, I hope you're having a wonderful day ^^

MikaelaLovett, 2024-03-26:
> @ojppe while I think that would be a useful app, I don't think that will fit within Trail Sense's use cases

I think it might be helpful for people with hearing disabilities, if it could also detect things like the sound of water, thunder, cars etc. Or maybe even recognise what kind of sound it is, I'm not sure if that's possible? ^^ But I completely understand the microphone permission thing and that it maybe isn't necessary for survival. 🤔 And it might be too difficult to create. :// (Also, I use earplugs often due to my sensory sensitivities and often can't hear certain sounds that aren't as loud, but might be important, so this would be another case of use as well?)

kylecorry31, 2024-03-26:
> I think it might be helpful for people with hearing disabilities, if it could also detect things like the sound of water, thunder, cars etc. Or maybe even recognise what kind of sound it is, I'm not [...]

Hm, that's very interesting - I will need to give more thought into the sound identification portion (or sound amplification of certain frequencies) - it would be tricky, but is doable. I could see that being useful in a hiking situation (getting closer to cars/waterfall/thunderstorm).

kylecorry31, 2024-03-26:
> Hi there, as I wrote in my Google Play review, I really like this app, thank you so much for working on it! ✨ I was wondering if adding local photos (and perhaps notes or files) to the packing list [...]

Thank you! I think that's a great idea, I'll plan on adding that in a future release: https://github.com/kylecorry31/Trail-Sense/issues/2297

CallamJ, 2024-05-30:
As I continue to bring in packing lists into Trail Sense for later I have found some more features I would like to see.
Considering last time you ended up splitting my post up into separate posts anyway I will just bunch them all together here.

### More annotation
Giving packing lists descriptions would be very helpful.
Add an "optional" checkbox to an item's details to specify that it is not necessary to bring. These wouldn't contribute to the packed percentage and would be visibly shown to be optional somewhere in the packing list.

### Custom Categories
Ability to add/remove custom categories. Then items could be added to these categories. "Other" would be a default category if the item's old category is deleted.
Tapping on an item's category in the packing list should open the dropdown to select a new category. This is mainly to make reassigning categories easier.

kylecorry31, 2024-05-30:
> As I continue to bring in packing lists into Trail Sense for later I have found some more features I would like to see. Considering last time you ended up splitting my post up into separate posts [...]

Thank you for the suggestions, here are some issues to track these features:
- https://github.com/kylecorry31/Trail-Sense/issues/2409
- https://github.com/kylecorry31/Trail-Sense/issues/2410
- https://github.com/kylecorry31/Trail-Sense/issues/730
- https://github.com/kylecorry31/Trail-Sense/issues/2411

Darin755, 2024-06-08:
So I have been thinking and it might be possible to put an small LLM inside trail sense. I have been playing around Phi3 mini and it fits in a fairly small amount of space.

https://ollama.com/library/phi3

You could add Llama.cpp inside the app and then ship a AI assistant that works completely offline. What would be even more cool would be tweaking the model to be good at outdoor and camping related tasks. For instance, it could be tweaked to do plant recognition and cooking. You could process images with llava as llava can use any text based model. 

https://github.com/ggerganov/llama.cpp

https://github.com/haotian-liu/LLaVA

Of course there would be some serious potential issues. First off, if the app started shipping a language model it would make the install size be huge. Even with a model under 2.5gb it still would be taking up a lot of space. You could have a button to download it after the app is installed but I think that would violate the no internet access policy. The other issue would be ram. Newer phones have plenty of ram so they will run it just fine but I have a 2019 phone with 3gb of ram. Theoretically it could work but it would be very tight. Running something so computationally intensive would also drain battery.

So in summary this would be cool but it would also be very hard to get right. I don't think this is immediately viable but I figured I'd drop it here as something to consider. Thank you for your time and development of this app.

Saijin-Naib, 2024-06-08:
I am incredibly wary of taking safety-critical instructions from a ML model, given how they work. 

In that respect, I am not sure how well it fits in with the rest of the very carefully vetted tools. 

Is there a specialized problem space you envision it for?

kylecorry31, 2024-06-08:
> So I have been thinking and it might be possible to put an small LLM inside trail sense. I have been playing around Phi3 mini and it fits in a fairly small amount of space.
> 
> https://ollama.com/library/phi3
> 
> You could add Llama.cpp inside the app and then ship a AI assistant that works completely offline. What would be even more cool would be tweaking the model to be good at outdoor and camping related tasks. For instance, it could be tweaked to do plant recognition and cooking. You could process images with llava as llava can use any text based model.
> 
> https://github.com/ggerganov/llama.cpp
> 
> https://github.com/haotian-liu/LLaVA
> 
> Of course there would be some serious potential issues. First off, if the app started shipping a language model it would make the install size be huge. Even with a model under 2.5gb it still would be taking up a lot of space. You could have a button to download it after the app is installed but I think that would violate the no internet access policy. The other issue would be ram. Newer phones have plenty of ram so they will run it just fine but I have a 2019 phone with 3gb of ram. Theoretically it could work but it would be very tight. Running something so computationally intensive would also drain battery.
> 
> So in summary this would be cool but it would also be very hard to get right. I don't think this is immediately viable but I figured I'd drop it here as something to consider. Thank you for your time and development of this app.
> 
> Edit: you also could use Tinyllama
> 
> https://ollama.com/library/tinyllama
> 
> https://github.com/jzhang38/TinyLlama

Hi, thank you for the suggestion. While it would be interesting to have an offline hiking-related LLM, I agree with all @Saijin-Naib 's concerns.

While I don't plan to add it to Trail Sense as a tool, if there's an open source app which has an offline LLM that you think would be useful, I can evaluate it an potentially list it as a recommended app if it performs well, has appropriate disclaimers, and is useful in the wilderness. Personally, I prefer the Offline Survival Manual for any information regarding wilderness survival - but I don't have a good recommendation for a plant ID app (I use Seek, but it isn't open source - and doesn't always perform well).

Saijin-Naib, 2024-06-08:
PlantNet is a FOSS tool and database for plants I use which has offline capabilities.

kylecorry31, 2024-06-08:
> PlantNet

Thank you, I will take a look!

Darin755, 2024-06-09:
I think it would be useful for completing a variety of tasks. For a safety perspective you could simply have a disclaimer with a warning about bad information.

I already run a model on my laptop which works pretty well. I connect my phone to my laptop. It is pretty much a glorified search engine that can create things from scratch. For instance, the other day I asked it to give me ideas for using some left over ingredients. I ended up making a curry recipe that it suggested. I had to slightly tweak it but it worked fine. There is very much a danger of bad information so you do need to be careful.

I am not that surprised that you are not really interested in adding this. Maybe this could be something to reevaluate after the LLM tech as had a chance to cool down a bit.

kylecorry31, 2024-06-09:
> I think it would be useful for completing a variety of tasks. For a safety perspective you could simply have a disclaimer with a warning about bad information.
> 
> I already run a model on my laptop which works pretty well. I connect my phone to my laptop. It is pretty much a glorified search engine that can create things from scratch. For instance, the other day I asked it to give me ideas for using some left over ingredients. I ended up making a curry recipe that it suggested. I had to slightly tweak it but it worked fine. There is very much a danger of bad information so you do need to be careful.
> 
> I am not that surprised that you are not really interested in adding this. Maybe this could be something to reevaluate after the LLM tech as had a chance to cool down a bit. Right now there are no known apps that do this which makes sense as it is very hard to run locally for the reasons I listed above.
> 
> Thanks for your time

Thank you for providing more information - I will consider adding it as a plugin if LLMs (or whichever technology ends up replacing them) get more reliable and are shipped on more phones out of the box.

Saijin-Naib, 2024-06-26:
I would love to see, if possible, the ability to use sensor fusion to enable GPS+Magnetometer blending for heading.

In many cases (especially on lower-end devices), it can be much more robust and stable to have a GPS-based heading, especially once you have a few seconds of movement.

ArduPilot uses this in the EKF3 filter as a settable parameter, if that helps when investigating this:  
https://github.com/ArduPilot/ardupilot

Saijin-Naib, 2024-06-26:
I would love to see, if possible, a way for Photo Maps and Beacons to become more integrated.

Right now, I have downloaded and calibrated trail maps for the County where I live. I also collect Beacons for each Feature on the map that I encounter.

It would be excellent if I were able to view the Beacons in the same location as the Photo Map they are associated with, so I can say oh, Shelter Orange is part of Park Blue, and is associated with the Features map.

Saijin-Naib, 2024-06-26:
I would love to see, if possible, filter results for Beacons and Photo Maps also match/highlight on the Group/Path for the object.

This would make a result for a PhotoMap named Trails under ```Country\State\County\Municipality\Park Blue``` when I search for Blue.

Currently, I have to name my PhotoMaps something like Trails - Park Blue to find them by where they belong to.

Saijin-Naib, 2024-06-26:
I would love to see the ability to have a ReadMe.MD at the root level of a Group in Photo Maps that could contain whatever local-only markdown content (so no remote images). 

One could use this to hold information about that Park or Trail that isn't on the Photo Map itself, for instance.

NextCloud Files handles this pretty neatly, though it could just be hidden under a collapsed header at the title or a tab, perhaps.

kylecorry31, 2024-06-26:
@Saijin-Naib thank you for the suggestions. Here are some issues which I think summarize all of them:
- #2455
- #2456, #1300, #1254
- #2457
- #2458

Please let me know if I got any of those wrong.

maverick74, 2024-06-27:
I think it would be great to have, in Astronomy, the Solicitists and the Equinoxes dates marked 😀

kylecorry31, 2024-06-27:
> I think it would be great to have, in Astronomy, the Solstices and the Equinoxes dates marked 😀

Thank you for the suggestion. Here's an issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/2460

maverick74, 2024-06-28:
> Thank you for the suggestion. Here's an issue to track that: #2460

Thanks :)

Saijin-Naib, 2024-06-28:
I would love to see the ability to see Estimated H/V accuracy while recording Points and Tracks.

Saijin-Naib, 2024-06-28:
I would love to optionally have the ability to record points using GPS Averaging/long baseline sampling to improve accuracy of recorded points, especially in challenging areas. 

UX could be, when enabled, tap to sample/stop, long press to instant mark. 

It would require preferences to enable, the desired accuracy to stop on, the desired max sample time/count (1/s assumed).

kylecorry31, 2024-06-28:
@Saijin-Naib thank you for the suggestions
- Can you clarify how you would want to see the horizontal / vertical accuracy of beacons and paths? Do you want to see them on the radar compass / map or just see the accuracy as text?
- I believe the feature you are looking for is GPS Smoothing, this is experimental, so you can enable via Settings > Experimental > GPS Smoothing. It works by averaging the readings (using a kalman filter), so the GPS will "improve" over time. I say improve in quotes, because from my research Android already performs GPS smoothing under the hood. As for the accuracy stop, here's an issue for that #2461

Saijin-Naib, 2024-06-28:
- Accuracies: Just as text when recording or reviewing them, I think. On the radar or AR, inaauracy circles might feel cluttered? 

- Smoothing: Almost, yes, but specifically for recording Points, like how GPS Point does, which significantly improves accuracy if you can wait. 

https://gpspoint.app

Garmin mapping-grade GPS also can do this, and it rocks.

kylecorry31, 2024-06-29:
> * Accuracies: Just as text when recording or reviewing them, I think. On the radar or AR, inaauracy circles might feel cluttered?
> * Smoothing: Almost, yes, but specifically for recording Points, [...]

Thank you for the clarification.

I agree, it would be too cluttered to show accuracy on a map view. Here's the issue to track that: #2463.

Thank you for the examples, I'll take a look. Here's an issue to track that #2464

Saijin-Naib, 2024-07-01:
It would be amazing if the import of PDFs for Photo Maps would have a control (scrub bar + direct numeric input?) for choosing which page to input from a PDF after switching to and previewing it. 

User story:  
Municipalities here often have multipage park feature and trail maps that do not always have the map on the first page, necessitating using other tools to split out pages from the PDFs.

kylecorry31, 2024-07-01:
> It would be amazing if the import of PDFs for Photo Maps would have a control (scrub bar + direct numeric input?) for choosing which page to input from a PDF after switching to and previewing it. [...]

Thank you, this will be implemented in #913

Saijin-Naib, 2024-07-20:
I would love to be able to jump to PhotoMaps that I am currently navigating on, or located on, from the compass, like how we can get to Beacons with the UI button.

Saijin-Naib, 2024-07-20:
I would love a way to refine calibration on PhotoMaps after initial calibration. 

Say we are navigating on the PhotoMap and notice our location is off from the map, we could enter a refine mode where we drag/drop our current position marker to the right spot and the projection gets transformed to better fit.

kylecorry31, 2024-07-20:
@Saijin-Naib thank you for the suggestions 

1. Have you tried the "Open Photo Map" quick action? I've been trying to prefer quick actions over additional UI elements since they are more customizable.
2. Would this be different from the ability to recalibrate a photo map (menu > calibrate)? Unfortunately, photo maps only supports 2 calibration points, so if you are looking to have a non-linear warping, then I'm afraid I may not be able to support that

Saijin-Naib, 2024-07-20:
Hi! 

1. Yes, I have it on the bottom, but I need to drill down to the map(s) each time through my organization structure (Country/State/County/Town/Park/Maps) and find the map. I have 36 maps currently, adding more each week. 

2. Hmm... in that case, I guess it must not be different if we are only able to do two-point...

kylecorry31, 2024-07-20:
> Hi!
>
> 1. Yes, I have it on the bottom, but I need to drill down to the map(s) each time through my organization structure (Country/State/County/Town/Park/Maps) and find the map. I have 36 maps [...]

Just to confirm, there are two photo maps quick actions - one to open the photo maps quick action tool (Tool: Photo Maps), and another to open the active map (Open Photo Map), which factors in your current location and the destination beacon. Can you confirm you are using the "Open Photo Map" variant of the quick action?

Saijin-Naib, 2024-07-20:
Ah, no, I am using the bottom navigation area to switch between compass and photo map:  
[image]

[42 more replies]

closed 2024-10-20 by kylecorry31