kylecorry31/Trail-Sense #1920: Sky map

kylecorry31, 2023-08-13:
Like the radar compass, but for the sky (both night and day).

The can be the precursor to #1809 as it is much easier to implement - and it may be able to be kept in place (with a view in 3D button).

See #1809 for details on what stars/planets to show.

Add a seekbar to change the time of day (from sunset the day before to sunrise the next day), and a calendar to change the date

michaelblyons, 2024-12-02:
For the watchers, some recent commits are interesting; https://github.com/kylecorry31/Trail-Sense/commit/c79b821c2d74d53ab4936b1b19a1e349d62468f1 for instance.

kylecorry31, 2024-12-02:
> For the watchers, some recent commits are interesting; [c79b821](https://github.com/kylecorry31/Trail-Sense/commit/c79b821c2d74d53ab4936b1b19a1e349d62468f1) for instance.

Yup, I started implementing the star catelog for fun :)

In the next release, at night the AR tool will show the top 50 stars + 57 navigation stars (there's a lot of overlap) and a handful of stars in major constellations used for navigation (many of these additions happened behind the scenes in https://github.com/kylecorry31/sol).

I've also been experimenting with #1919, and want to make that happen soon (still verifying the accuracy in the best case scenarios to see if it will even be feasible though). If I can't get that to an accuracy of meters, I'll likely just put it in another app.

kylecorry31, 2025-12-24:
I'm going to close this out as not needed since the augmented reality tool is the better fit for this. If people do really want a 2D version of that, please comment here with reasons you think it would be useful.