kylecorry31/Trail-Sense #1935: Experiment: Tool grid

kylecorry31, 2023-08-23:
The goal of this is to just experiment with a new layout for Trail Sense. I don't think I will switch it over, but I'll never know unless I try it out.

Idea:
- [ ] Remove bottom navigation
- [ ] Create a single tool grid (like an app launcher / home screen)
- [ ] Section at the top for frequently used tools (default to navigation, astronomy, weather)
- [ ] Section at the top for quick actions
- [ ] Display options: Alphabetical, Categorized, Custom (allow user to reorder grid)
- [ ] Figure out how to return to the grid (just pressing back several times or a persistent bottom bar to go back to all tools)
- [ ] In the future, this layout could support widgets

If this experiment works out well, I will consider adding this to Trail Sense (at least an experimental feature).

kylecorry31, 2023-11-11:
Proof of concept:

[image]

kylecorry31, 2023-11-11:
The first version of this will be available in tonight's nightly build

closed 2023-11-13 by kylecorry31

michaelblyons, 2023-12-02:
This is great, btw. 🎉

kylecorry31, 2023-12-02:
> This is great, btw. 🎉

Thanks! This change opens up a lot more possibilities, especially with #2046 and #2074

commit e4752987 (2023-11-13): Add notice of settings move

Closes #1935