kylecorry31/Trail-Sense #1950: GPS stuck on searching, except when I'm running GPSTest at the same time.

Void-0000, 2023-09-03:
I noticed the app is never able to detect any kind of GPS signal, so I installed [GPSTest](https://github.com/barbeau/gpstest) to see if I couldn't figure out what the problem was... and then it started working, until I stopped GPSTest, at which point it instantly lost the signal again.

I've noticed this happens with certain other apps as well, for example [BLE Radar](https://github.com/Semper-Viventem/MetaRadar) but not with others, like [Organic Maps](https://github.com/organicmaps/organicmaps).
Maybe there are different ways of accessing GPS location?

I'm using the Nothing Phone 1 and have Magisk and MicroG, in case that affects it.

kylecorry31, 2023-09-03:
Hi, can you try the following:
1. Under Trail Sense Settings > Sensors > GPS, disable "Require satellite fix"
2. If that doesn't help, can you open Trail Sense Settings > Diagnostics and let me know if there are any issues reported

For reference, in the nightly builds I updated how Trail Sense determines how many satellites there are since some phones were not populating the satellite count directly (#1942). If you would like to see if this is fixed on the nightly build, that would be greatly appreciated. Details on the nightly builds can be found here: https://github.com/kylecorry31/Trail-Sense/discussions/1940

Void-0000, 2023-09-03:
Disabling "Require satellite fix" solved my problem, but the satellite count still displayed as 0, on the nightly build though everything worked correctly (GPS signal is found and satellites show up, both with and without "Require satellite fix").
Seems like it's the satellite count issue you mentioned then, thanks!

kylecorry31, 2023-09-03:
Glad that worked, and thanks for testing out the nightly build! It was hard for me to test that that new method myself since my device manufacturer populates the satellite count using the old method as well, so this was perfect timing.

closed 2023-09-03 by kylecorry31