kylecorry31/Trail-Sense #1953: Re-evaluate / remove cliff height tool

kylecorry31, 2023-09-04:
I'm not sure if this tool actually has a valid use case and a lot of people find it confusing.

You could use it to answer the question of how far down is a drop - but what does that actually provide? If you need to lower something down with a rope, you can just use the rope to measure distances.

Other than that, I'm not seeing any use cases. This tool was added at a time when I didn't have a clear goal for Trail Sense, but now that I do it may be time to retire this tool.

I'll leave this issue up until after the next release in case anyone can provide a valid use case for it.

kylecorry31, 2023-09-07:
In the October release, I will mark this tool as experimental - so it will still be available. Upon opening I will put a notice that this tool will be removed and leave it there until December. The notice will also ask users to provide me feedback on if they are actually finding this tool useful within Trail Sense's core use cases.

Depending on how that goes I will either keep the tool in, adjusting it to better support use cases presented, or remove it.

cziss, 2023-09-16:
Hi, I find that tool useful. You are right with the rope, but then you have to bother with packing and unpacking rope. That can be problematic in cases when time is important for you. And also, I like it for just "fun" to find out depth under the seeing or bridge... :)

kylecorry31, 2023-09-16:
@cziss thank you for the details. Can you clarify if there are any other use cases you've used it for other than measuring the height of cliffs/bridges for fun?

I'm hoping to identify other use cases than laying down rope - though if that's the only practical use case, and people are actually using it for that, I'll leave it as an experimental tool and potentially rebrand it in the future.

cziss, 2023-09-16:
Well, for main user case - laying rope. And another to measure if actual height is safe for jumping into water (over cliff or bridge). Of course before inspection conditions in water (visual, diving...).

kylecorry31, 2023-09-16:
@cziss thank you for clarifying

kylecorry31, 2023-11-27:
It appears that several people are using this for rock climbing to measure the rope. While rock climbing isn't officially supported, I think I will keep this tool in Trail Sense, but it will need to be enabled within the Experimental settings rather than keeping it on by default. I'm not a big fan of having to drop an object down the cliff to measure the height, but I don't see an alternative at this moment.

I would also like to add instructions to the tool itself, similar to the Water Boil Timer - and also reminding the user to be 100% sure it is safe to drop an object to measure the height.

cziss, 2023-11-28:
Hi, I think that it's best solution. Thx.