kylecorry31/Trail-Sense #1960: Cell Signal Dot not showing anything

shortpupper, 2023-09-10:
There should be a grey dot if the signal can't be found, or red if poor aka 1 bar out of 5. Pictures showing what I mean by the election dots are there but then some of the cell signal dots aren't. Android 13, version 5.2. [image]

[image]

shortpupper, 2023-09-10:
Elevation, not election

kylecorry31, 2023-09-10:
Thank you for the suggestion - I'll experiment around with displaying a gray circle when there is no signal.

My reasoning for not displaying anything when there is no signal was to make it really clear where there was a signal and where there is nothing.

The red dots in the screenshot above represent a poor signal, so if I do implement this, I am leaning toward using a shade of gray.

shortpupper, 2023-09-10:
Sorry for the confusion on the 1 to 5 bars red, When I would do this I would go through a no-bar area but it would still show on my phone as 1 bar.

kylecorry31, 2023-09-11:
Hi, just to confirm you are saying that Trail Sense shows your phone as 1 bar when you actually don't have signal, or that your phone is saying 1 bar when you don't have signal?

If it is that your phone is showing 1 bar, I wonder if that's because your carrier has no service, but it still has service with another carrier (for emergency services).

Trail Sense uses the following classification on a 3 bar system.
- No signal (0 bars)
- Poor signal (1 bar)
- Moderate signal (2 bars)
- Strong signal (3 bars)

shortpupper, 2024-07-03:
Apologies for the very long wait asfor a response. My device has a set of 5 bars for signal/mobile data. Thank you very much for your time as well.