kylecorry31/Trail-Sense #1961: Compatibility with BLE LoRa Communicators

LolloDev5123, 2023-09-11:
Add an experimental feature that allows TS users to connect their device to a BLE mesh off-grid communication device like https://meshtastic.org.
There's already an android app for meshtastic, but it'd be very cool to see intertwined TS features like sensor readings, backtracking, weather, waypoints, chat, etc. incorporated with meshtastic (see meshtastic telemetry features, this allows for "remote" barometers, all connected by mesh)

[image]
[image]
[image]

kylecorry31, 2023-09-11:
Thank you for the suggestion, I will consider adding this along with https://github.com/kylecorry31/Trail-Sense/issues/1540 and https://github.com/kylecorry31/Trail-Sense/issues/1534

msherba, 2023-09-12:
Meshtastic app offers the imeshservice to interface with the hardware devices. For an example of a plugin using it see the Meshtastic ATAK-Plugin which takes CoT messages from ATAK and sends them over the Meshtastic device paired via Bluetooth in the MT app.

kylecorry31, 2024-10-25:
Moved to https://github.com/kylecorry31/Trail-Sense/issues/1540 to tidy things up for now

closed 2024-10-25 by kylecorry31