kylecorry31/Trail-Sense #1965: Translation error on weblate

Craftefix, 2023-09-16:
Hello i cant edit the german weblate sentences because nobody can save the Translation but german got only 97% translate because of the new update

kylecorry31, 2023-09-16:
Hi, Weblate appeared to be having some issues this morning (I was getting 404s trying to load the page).

I'm not sure if there are still any ongoing issues on their end, but I've updated the Trail Sense repo there and translations appear to be unlocked. Can you try again?

Craftefix, 2023-09-16:
[image]

Look i cant click that button to save and continue and the button for save and stay is completly missing

kylecorry31, 2023-09-16:
Hi, the app_name is marked as not translatable. That is intended, and was marked as such a few years ago.

kylecorry31, 2023-09-16:
Here's what I have for filters:
[image]

kylecorry31, 2023-09-16:
So looks like the filter should be: state:<translated instead of state:>=translated

Craftefix, 2023-09-16:
oops, now i can translate more, thanks for your help

closed 2023-09-16 by Craftefix

kylecorry31, 2023-09-16:
Thank you for providing translations!