kylecorry31/Trail-Sense #1976: Storm, Storm and again Storm Warnings

sleeplessnight2, 2023-09-25:
I keep getting storm warnings. This is especially bad when you are driving in the mountains.

There seems to be no alignment with the GPS position, so that one can assign a strongly changed pressure to a position change and thus no alarm is triggered.

even this morning is displayed to me complete crap, although I was also time only @home. (see picture)
[image]

Should the cause be a calibration, an automatic "mute" function of an alarm is also missing here.

Other apps or even e.g. Garmin can do this much more reliably, although they have to deal with the same problem.

kylecorry31, 2023-09-25:
Hi, I'm sorry you are having issues with Trail Sense.

Trail Sense is only designed for use in hiking / outdoor activities. I know some like to turn the weather monitor on before leaving for a trip so they don't forget to enable it when they start their hike, so I'm planning on implementing #1858 which should eliminate the false alerts while you are driving.

In case it also applies here, please see this discussion on using Trail Sense indoors / while driving (both are unsupported, and can lead to inaccuracies - especially if the building is large like it was in the referenced issue): https://github.com/kylecorry31/Trail-Sense/issues/1833

As for the alert happening repeatedly - is that occurring every 15 minutes for you? Trail Sense has a mechanism in place where it won't send the alert if a storm alert was previously sent. It resets that trigger when a storm is no longer predicted. So I wouldn't really expect that to occur back to back (I have not seen that behavior on my end). I'll do more investigation to make sure that is working properly though, and maybe also put a minimum time in there (though I need to be careful that it doesn't miss a subsequent storm).

kylecorry31, 2023-10-08:
Closing this issue in favor of https://github.com/kylecorry31/Trail-Sense/issues/1858 and https://github.com/kylecorry31/Trail-Sense/issues/1833

closed 2023-10-08 by kylecorry31