kylecorry31/Trail-Sense #2010: Improve messaging around missing compass (again)

kylecorry31, 2023-10-22:
Some users are still not understanding that there's nothing that can be done if their device does not have the hardware, and think I'm trying to sell something.

I don't know how to make the messaging any clearer, but maybe the problem is they are skipping through the tutorial without reading, see the banner for the no compass and aren't reading it. 

Idea
- [ ] Show no compass dialog on first load of the navigation tool

If anyone has a suggestion for how I can make this clearer, please let me know:

Here's the current messaging:
**Title:** Your device does not have a compass sensor
**Description:** Your device does not have a built-in compass sensor. This is a hardware component, so there is nothing that can be done to add it. Some features that use the compass will be unavailable

This messaging is shown on the onboarding pages (separate page around no compass), the title is visible in the error banner on the navigation tool, and clicking that will show both the title and description.

kylecorry31, 2023-10-22:
Previous issue: #https://github.com/kylecorry31/Trail-Sense/issues/1758

JensKorte, 2023-10-29:
Either your phone has no compass included or it is broken. Please ask your phone seller for help.

You can install the app Offline Survival Manual and read the chapter [Field-Expedient Direction Finding](https://github.com/ligi/SurvivalManual/wiki/DirectionFinding) how to find north.

kylecorry31, 2023-10-29:
Thanks, good idea - I'll add to it that they should contact their phone seller for help.

kylecorry31, 2023-10-30:
An example of a recent review where this may have helped: "The compass doesn't work. It says that my device does not have a compass sensor. My device is current and not cheap."

It shows that the user believes their device should have the sensor and that Trail Sense is not detecting it. When I looked up the device on the manufacturer's website, it confirms there is no compass sensor.

Add the following to the compass dialog:

If you believe this is incorrect, you can search for "{device manufacturer} {device name} sensors" and see if magnetometer / compass is listed or you can reach out to {device manufacturer} to see why this sensor is not available.

JensKorte, 2023-10-30:
Does Android include an API to access the compass and the apps don't need to know details about the hardware and its drivers? Then you could stress, that TS cannot do anything about it, if android doesn't provide the access.

kylecorry31, 2023-10-30:
Yes, Trail Sense is using that API to determine if the device has a compass and displaying the message when it says it is not available. I tried to convey that TS can't do anything in the messaging, but based on the messages I've received after I pushed that change, it seems that some people don't believe it is correct because their phone is new and either don't know how to check online or don't want to (there's nothing I can do about the later). I'm sure the API is accurate, because every time I've looked up their device for them the manufacturer indicates there is no compass sensor.

I'll come up with a new message and post it here after - I want to keep the line about TS not being able to add hardware to the phone and I like adding in how to find details themselves if they don't trust TS.

kylecorry31, 2023-10-30:
Maybe also a better onboarding page will help. There's currently one that displays if you don't have a compass, but it's styled just like the rest of the pages, so maybe people are ignoring that.

Something like "Sensor Check" that checks to see if they have all the sensors that TS uses and displays that wording about manufacturers at the top, and a red X next to each missing sensor with a description of what will be unavailable.

kylecorry31, 2023-11-11:
Also, I should highlight the part about there being nothing that can be done to fix this in the message - I've had multiple people say I'm trying to sell them a compass sensor to make Trail Sense work.

kylecorry31, 2023-11-11:
## Original messaging
Your device does not have a built-in compass sensor. This is a hardware component, so there is nothing that can be done to add it. Some features that use the compass will be unavailable.

## Proposed update
Your {device name} doesn't come with a built-in compass sensor. **Since this is a hardware feature, there is nothing that can be done to add it.** If you're unsure or have questions about your device, you can check your device manufacturer's website or get in touch with them for more details. Unfortunately, this means that some features in Trail Sense that rely on the compass won't be available for you. If you need assistance or have any questions, feel free to reach out at trailsense@protonmail.com.

I may leave off the last line, because even though I want to be able to help them, I wonder if that will feel like I'm secretly trying to sell them something to fix it.

JensKorte, 2023-11-11:
I wouldn't include the mail adress. Instead I would add a second chapter like

### For those who still think I want to sell something extra

 No! There is no payed version ot Trail-Sense that has additional features! And Yes, there are new mobile phones on the market that are not cheap and have no compass included.

kylecorry31, 2023-11-11:
> I wouldn't include the mail adress. Instead I would add a second chapter like
>
> ### For those who still think I want to sell something extra
> No! There is no payed version ot Trail-Sense that has [...]

Thanks, so maybe I should include an FAQ section there, I like that.

JensKorte, 2023-11-11:
> Thanks, so maybe I should include an FAQ section there, I like that.

That could work, at least for many people. ;)

kylecorry31, 2023-11-11:
Here's some updated messaging that I think addresses the most common questions. I left out even mentioning that some people think I'm looking to sell them something, hoping the "is there a way to fix this?" section makes that clear:

----------------

Your {device name} doesn't come with a built-in compass sensor. Unfortunately, this means that some features in Trail Sense that use the compass won't be available for you.

## Is there a way to fix this?
No, unfortunately this is a hardware feature, so there is nothing that can be done to add it. All other compass apps will have the same issue.

## My device is new/expensive, why is it missing this sensor?
Some phone manufacturers choose not to add this sensor, even on new or expensive devices. You can get more information by visiting your device manufacturer's website or getting in touch with them.

JensKorte, 2023-11-11:
That is more polite.

commit 7d1c76c8 (2023-11-17): Update compass messaging

Closes #2010