kylecorry31/Trail-Sense #2024: Augmented reality tool

kylecorry31, 2023-10-29:
- [ ] #608 
- [ ] #2023 
- [ ] #1702 
- [ ] #1802 
- [ ] Toggle on / off layers
- [ ] Support linking from other tools with context (ex. astronomy would disable beacon layers by default and show a time slider)
- [ ] Support for choosing guidance targets (ex. sun, moon, constellation, navigation beacon, etc) - displays as a bottom sheet categorized by type (astronomy, location)
- [ ] Support for time slider (changes the current time of the view - affects astronomy objects)

michaelblyons, 2023-10-29:
This seems extremely cool. Are there device limitations or external apps (ARCore?) required for the AR stuff to work?

kylecorry31, 2023-10-29:
@michaelblyons there shouldn't be any limitations for this to work - other than your phone needs a compass/accelerometer for it to match the rotation of the device (though there will be a version that you can manually drag) and a back facing camera if you want the full AR experience. A gyroscope would be helpful too, but it can work without that (just won't be as accurate / responsive).

I'm building it from scratch, so it doesn't depend on any external apps.

The only thing I'm not sure about is if all devices report camera field of view - the docs imply that it does, but I'm not sure I trust that because I've seen other things that some manufacturers just ignore. I may add a manual calibration option if I find this to be the case (or maybe add it just in case)

This tool is available on the Nightly builds if you wanted to try it out: https://github.com/kylecorry31/Trail-Sense/discussions/1940

Turn on Settings > Experimental > Augmented Reality, then open the Tools > Augmented Reality. It's super early (as in I built it within the last couple days), so I'm sure there are bugs / unoptimized or approximate algorithms. Currently it will display the sun / moon positions and nearby beacons. I also noticed it is off by about a degree on my device, which may be the maximum accuracy I can get on a phone without some fancy computer vision algorithms (I was thinking of auto-calibrating using the actual sun/moon), but haven't researched that much.

I would love to get feedback on this tool and hear ideas for what I can include in it (feel free to post ideas here). You can see some planned features by using the label:augmented-reality in the issue search.

closed 2024-04-20 by kylecorry31