kylecorry31/Trail-Sense #2025: Adjust AR calibration using sun / moon

kylecorry31, 2023-10-29:
Make slight modifications to the AR calibration in real time if the sun / moon is visible in the camera frame, but not 100% aligned with the AR position.

Calibration can include:
- FOV calibration
- Compass / clinometer calibration

Note: This may also be a viable option for manual calibration - point phone toward the sun and rotate it a little to estimate FOV.

This can be a setting, and based on my testing maybe it can save the results.

kylecorry31, 2023-12-28:
You should choose the object to calibrate with (sun or moon only for now - but in the future a star or beacon can be used), then choose how you want to calibrate (auto, manual).

If auto is selected it should have a slider for the threshold and draw an indicator on the detected object. 

If manual is selected the user will be asked to center the object on the screen.

Once the user is satisfied with the selection, they can click "Calibrate" to save changes.

Calibration is reset on pause (maybe?)

michaelblyons, 2024-02-17:
Is this like Phone = Sextant?

kylecorry31, 2024-02-17:
Sort of.

Because the model of sun/moon motion is so accurate (to a fraction of a degree), it can be used to calibrate for the inaccuracies in the magnetometer. The phone's sensors aren't accurate enough to get your location using the sun's position, so not entirely replacing the purpose of a sextant. Rather this is a way to provide a temporary offset for magnetic anomalies and noise (ex. switch to the gyro once calibrated). 

In the future the accelerometer can be calibrated this way as well, but not a high priority since it is inherently more accurate than the magnetometer because the only consistent force acting on it is gravity.

The only downside (other than being annoying to do) is the calibration is only accurate for a short amount of time because the gyro/magnetometer will drift by a few degrees. Hopefully I can implement an automatic calibration which recalibrates whenever the sun/moon is visible, which will help out a bit.

Normally this inaccuracy doesn't matter because it's hard to point your phone with that much accuracy as a regular compass (ex. for navigation), but because AR is drawing on the word it is noticeable over long distances (like the sun/moon for example), where even being off by a degree will add up to being noticeably off.

closed 2024-02-18 by kylecorry31