kylecorry31/Trail-Sense #2120: [Tracking Issues] More Quick Settings Tiles

kylecorry31, 2023-12-15:
- [ ] Flashlight
- [ ] Screen Flashlight
- [ ] Whistle
- [ ] Sunset alerts
- [ ] White Noise

Saijin-Naib, 2024-01-27:
It looks like v5.7.1 notes say Flashlight tile should be live, but I cant seem to find it in my F-Droid build on /e/OS v20.0.0 (Android 12). 

Device compat. Issue?

kylecorry31, 2024-01-27:
> It looks like v5.7.1 notes say Flashlight tile should be live, but I cant seem to find it in my F-Droid build on /e/OS v20.0.0 (Android 12).
> 
> Device compat. Issue?

Hi, no the flashlight quick settings tile (system notification shade) was not yet implemented. The feature that was released in 5.7.* was the Screen Flashlight quick action (the buttons in the top corners on Trail Sense / on the top of the tools tab).

Saijin-Naib, 2024-01-27:
Oh, sorry for the noise! That works lovely