kylecorry31/Trail-Sense #2130: Settings user guides

kylecorry31, 2023-12-20:
- [ ] General Settings (Units, Theme, Power, Privacy, Errors, Backup / Restore)
- [ ] Sensors > Compass
- [ ] Sensors > GPS
- [ ] Sensors > Altimeter
- [ ] Sensors > Barometer (rewrite)
- [ ] Sensors > Thermometer (rewrite)

Basically the same as the tool guides in https://github.com/kylecorry31/Trail-Sense/issues/2074

VeslaZ, 2024-01-06:
I was thinking this morning about an AI assisted tool that could work with
the GPS locator. It would recognize location and give information based on
that location. I.e. animals, foliage, poisons, etc. it would generate
important information about the location and give recommendations based on;
season, etc.
Kind regards,
Will

kylecorry31, 2024-01-06:
@VeslaZ thank you for the suggestion - I have considered a similar topic with #459 for tool recommendations. Unfortunately more detailed information such as animals, plants, geography, etc would not be feasible without using the Internet (or taking up a lot of storage space). It could be a good idea for a plugin though (#1540)

Also, for future reference I recommend posting to #1911 instead since that is out of scope of this issue (settings user guide).

closed 2024-04-20 by kylecorry31