kylecorry31/Trail-Sense #2156: Allow changing the size of the chevron in the photo maps

jquillan, 2024-01-11:
The chevron is quite small and hard to see, epically in bad lighting conditions. If this could be larger (and maybe allow color change as well) this would be helpful.

Increasing the point size on paths might be a nice feature as well, but not as critical.

kylecorry31, 2024-01-11:
Hi, thank you for the suggestion. Just to confirm, you are referring to the arrow that indicates your location, right?

jquillan, 2024-01-12:
> Hi, thank you for the suggestion. Just to confirm, you are referring to the arrow that indicates your location, right?

Yes,
Here is a screenshot of what I am referring to
[image]

closed 2026-03-02 by kylecorry31