kylecorry31/Trail-Sense #2159: Triggers and actions

kylecorry31, 2024-01-13:
While writing up #2157 I thought it may be useful to let the user trigger behaviors for other things such as time of day, location (geofence), distance travelled, etc. This could also be the foundation of a generic alert mechanism (#1915, #719, #647).

This could be a tasker/IFTTT like feature where the user specifies the trigger and a list of actions that should occur. 

More research is needed to see if this would be helpful and what existing alerts could be replaced by it (and also allow pre-made actions to be set up for things like the sunset alert).

kylecorry31, 2024-01-13:
Even if the first iterations aren't user facing, this would be nice to have from a development perspective. 

A broadcast receiver that takes in an action ID and executes the associated action(s).

kylecorry31, 2024-01-13:
Triggers
- Sunrise/set/noon (type and offset)
- Time
- Timer (countdown)
- Tide level (high, low, tide table)
- Geofence (enter, exit)
- Power saving mode (enabled, disabled)
- Device battery is low
- Device charging state (start charging, stop charging)
- Backtrack (enabled, disabled)
- Weather monitor (enabled, disabled)
- Pedometer (enabled, disabled)
- Astronomy event (meteor shower, eclipse)
- Distance travelled (currently the distance alert - specify source?)
- Cell signal state changed (established or lost - from backtrack)

All triggers have the ability to repeat on the next occurrence (indefinitely). Triggers also can send data to the actions (type: number, text, datetime)

Actions (multiple allowed - executed in sequence):
- Set a trigger (ex. timer)
- Change backtrack state (enable, disable, frequency)
- Change weather monitor state (enable, disable, frequency)
- Change pedometer state (enable, disable)
- Send notification (type, title, body, icon - some of these will not be user changeable)
- Change power saving mode state (enable, disable)
- Sound off alarm (need a way to specify this in the trigger for the time based ones so they can be scheduled accordingly)
- Cancel a trigger

michaelblyons, 2024-01-13:
If you add Bluetooth permissions in the future, connect/disconnect of devices would also be a good trigger, #1223. If you don't want to add permissions, that's also reasonable.

kylecorry31, 2024-01-13:
Good idea, thanks!