kylecorry31/Trail-Sense #2175: Request: Update app icon

kylecorry31, 2024-01-21:
Update the app icon to be more modern (suggestion was to make it brighter and more attractive).

Not sure if I'm going to do this, since I like the simplicity of the icon - but I want to consider some alternatives before just closing this.

michaelblyons, 2024-01-26:
I use the Arcticons pack, so maybe I shouldn't weigh in, but I like what you have already. 😉

CloneWith, 2024-02-05:
the current one is good enough 👍

maybe better with the compass pointing to two corners of the icon, something like this:

[image]

LeftyDextrous, 2024-02-05:
I think the current icon is good, although it occurs to me that it may not distinguish the app from other single-function compass apps.

kylecorry31, 2024-02-05:
Thank you for the suggestions / feedback. 

It's true that it blends in with other single function compass apps, so if I do change the icon, I'll probably try to incorporate multiple tools into the icon. But, I've come to like the simple icon of Trail Sense, so I'm not sure I want to change it (plus I have a t-shirt for Trail Sense with that icon on it 😆)

kylecorry31, 2024-05-04:
After further consideration, I will not be changing the icon right now

closed 2024-05-04 by kylecorry31