kylecorry31/Trail-Sense #2179: Theme color is not correct when Dynamic Color is enabled

WildManiac, 2024-01-24:
The Dynamic Color setting is not applying the correct theme color when enabled.

| On Device Theme Color | Dynamic Color in App |
|---|---|
|[image]|[image]|

Trail Sense Version: 5.7.1-dev
Device: Samsung Galaxy A52 5G
Operating System: OneUI 6.0 (Android 14)

Please let me know if any other information is needed.

WildManiac, 2024-01-24:
"Most" of the apps that I have that have Dynamic Color configuration do respect the purple-ish color pallette. Below is what it looks like in Barcode Scanner:

[image]

Granted, Trail Sense isn't the first app that I have run into that doesn't apply the color pallette correctly, so I don't know if it is an app thing or a Samsung thing. But yeah, I can certain run a nightly whenever it is available.

kylecorry31, 2024-01-24:
I just tested in the Samsung remote lab, and it does respect the "Basic colors" theme just fine, but when you use "Wallpaper colors" it always uses the color in the middle as the primary color. I suspect there are other apps that do the same thing because that's the out of the box behavior with the Google Material library that's recommended. I'll do more investigation to see if I can make it use the top color in the wallpaper colors. 

I'll post here when a nightly fix is available.

WildManiac, 2024-01-24:
That is interesting, because no matter what colors I select, whether it be Wallpaper or Basic, Trail Sense still has the default blue theme when Dynamic Colors is enabled.

kylecorry31, 2024-01-24:
Hm, I tested on an Android 13 Samsung device, so many it's different across versions

kylecorry31, 2024-01-24:
Developer notes:

Looks like this is an upstream issue on the Google Material 3 library: https://github.com/material-components/material-components-android/issues/3924

Some people suggest downgrading to version 1.10.0 - I need to research the differences before I downgrade

kylecorry31, 2024-01-24:
@WildManiac a new nightly build is available with a fix for this (tested via the samsung remote device lab): https://github.com/kylecorry31/Trail-Sense/actions/runs/7647008723

Unfortunately, the issue was upstream in the Google Material library and I had to downgrade it to fix it. Unfortunately, I suspect until that is fixed more apps are going to start having this issue as they upgrade to the latest version.

WildManiac, 2024-01-25:
I can confirm that Dynamic Color is reflecting appropriately on my device.

[image]

And per your comment about other apps seeing this issue, that might be why I have some that don't reflect the appropriate color.

kylecorry31, 2024-01-25:
Thank you for testing that! I'll roll out a new release this weekend with the fix.

closed 2024-01-26 by kylecorry31