kylecorry31/Trail-Sense #2181: F-Droid is a few versions behind again

michaelblyons, 2024-01-26:
Do they not automatically pick up new versions?

kylecorry31, 2024-01-26:
Yes, all of that is automatic and on their end. It normally takes a week, but sometimes more. In the past I opened an issue after a week and a half and they just recommended waiting. I can look into opening an issue in a few days if it doesn't appear by then.

kylecorry31, 2024-01-26:
And I do see that they built version 5.7.1, just haven't updated the store page yet

michaelblyons, 2024-01-26:
Ah, okay. I had an update to a different app that came out after your 5.7.x series, so I thought there was something stuck.

But I see the build now. I suspect they have to do signing with human interaction, so the store update takes a bit after the build sometimes.

closed 2024-01-26 by michaelblyons

kylecorry31, 2024-01-27:
> Ah, okay. I had an update to a different app that came out after your 5.7.x series, so I thought there was something stuck.
> 
> But I see the build now. I suspect they have to do signing with human interaction, so the store update takes a bit after the build sometimes.

Just wanted to let you know that the F-Droid 5.7.1 version is finally available (5.7.2 was just released yesterday, so probably won't be in F-Droid for a bit) - not sure why that took so long (or why version 5.7.0 was skipped). There's a lot of behind the scenes stuff in F-Droid that isn't visible to developers unfortunately.

michaelblyons, 2024-01-28:
Oh, yes. I noticed and updated it already. I have been itching to try the AR stuff. It looks fantastic!