kylecorry31/Trail-Sense #2183: Users still not understanding the "no compass" messaging

kylecorry31, 2024-01-26:
I'm still receiving a lot of negative reviews / emails regarding the compass not working on devices without the compass. Some even mention the messaging that shows up. I'm at a loss of how to make this more clear.

On devices without compasses the following happens:
- The onboarding screen has a page dedicated to the missing compass sensor
- When you first open the Navigation tool, a dialog appears with details on the missing compass sensor
- The compass bearing is not shown on the Navigation tool (replaced with '-')
- Only the radar compass is available
- The compass is fixed to point to north
- A red banner is shown at the top of the screen that says "Your device does not have a compass sensor"
- The arrow on the radar compass is replaced with a circle

While I can't really see how this could be the case, my best guess is that some users are ignoring all the messaging and notice that the radar compass looks like a compass dial and therefore should work like one.

I would like to restrict Trail Sense in Google Play to just devices with a compass sensor, but I know there are people out there who understand the limitations of their device and are using Trail Sense for other things. It's just taxing to be getting a lot of negative messages over something outside my control.

What I will do:
- Remove the dial ticks and cardinal directions from the radar compass, and add a setting that only shows on devices without a compass for turning that on in Navigation settings
- Consider making the radar compass a square on devices without a compass, so it has no resemblance to a compass
- Completely remove the bearing title instead of displaying a '-'
- Remove the compass accuracy indicator
- Move the notice about compass sensor requirement higher up in the Google Play store listing (I don't think this will help, since I don't think the other messaging is being read)

In the future:
- #521 
- Don't have the Navigation tool pinned to the bottom nav by default (#2046)

michaelblyons, 2024-01-26:
That's a downer.

I'm sure it's more work to do so, but you might consider a second app with its own build target and yank any compass-required functionality. Then add a warning for no-compass users to install the other version, and eventually mark the full-function app as compass-only.

kylecorry31, 2024-01-26:
Hm, I hadn't considered that - I'll definitely check that out, but it may be too much work to maintain both (because I would also have to fill out the Google Play policies for both as well). Thanks!

kylecorry31, 2024-01-27:
I've made the changes outlined in the description, if that doesn't help then I don't know what will.

After further thought, I'm not going to create a second version of Trail Sense for non-compass devices. I'm just going to assemble a generic reply I can respond with and try to ignore the negativity.

closed 2024-01-27 by kylecorry31