kylecorry31/Trail-Sense #2273: Unreliable single tap on paths / photo maps

astewart77, 2024-03-16:
New user / new big fan!! New Kyocera Duraforce Pro 3 phone, so I don't know if this a phone problem or a Trail Sense problem.

All functions work as described, but a single tap to select a point on a path takes from 1 to 8 tries before it selects. Selecting a beacon on a photo map is a little better with 1 to 4 tries. Tap and hold works correctly on both. Double tap on a path always zooms, so it seems to see two discrete taps.

Tapping any other place in the UI works first time. I've tried Glove Mode and used a stylus instead of my fat fingers. That makes the rest of the UI more sensitive, but point selection is the same.

PS. I like this phone, but with your app, I'm wild about it. My old Garmin has retired to the junk drawer.

kylecorry31, 2024-03-16:
Hi, I'm sorry you are having issues. Both of those use "onSingleTapConfirmed" but also operate off of a point radius - so it could be either. It could also be something weird with that manufacturer, so I will probably require you to test any possible fixes I implement (this is the first report of that problem). I'll send instructions when I figure out possible causes.

Do you have the same issue with the Climate tool chart? And what about the astronomy chart (if you click the sun or moon)?

kylecorry31, 2024-03-16:
I believe I have resolved this issue - I made the click size larger and also keep it the same size (compared to the screen) when zooming in. It does seem a bit more reliable on my end as well.

I'll plan on including this in the release this weekend, but if you wanted to test it out before then, you can use the following nightly build: https://github.com/kylecorry31/Trail-Sense/actions/runs/8307227128

And here's a guide on how to install the nightly builds: https://github.com/kylecorry31/Trail-Sense/discussions/1940  (installs as a separate app, so it won't interfere with any existing data in Trail Sense, and you can uninstall the nightly version after you are done with it).

astewart77, 2024-03-16:
It works very well with a stylus. I still have fat fingers, so getting the correct point requires zooming way in. That's fine.
I had to look up how to allow unknown APK files. Will the new release show up on Google Play Store as an update?
Thanks.

kylecorry31, 2024-03-16:
> It works very well with a stylus. I still have fat fingers, so getting the correct point requires zooming way in. That's fine. I had to look up how to allow unknown APK files. Will the new release show up on Google Play Store as an update? Thanks.

I can look at increasing the tap size more, I'll experiment with it for the next release. 

And yes, the new release will show up on the Google Play Store as an update (should be rolling out now, so it may already be available)

closed 2024-03-16 by kylecorry31

astewart77, 2024-03-16:
I'm updated and working well. It always selects a point on the first try. With fat fingers, it's just not always the point I was aiming for because I'm overlapping multiple points. Zoomed in works great.

kylecorry31, 2024-03-16:
Thanks for trying that out. 

It's a little tricky with overlapping points since it's hard to tell which one is the correct one (it currently chooses the one that is rendered on top). I created a new issue to see if I can improve this for overlapping points: #2277