kylecorry31/Trail-Sense #2285: Sending GPS-Coordinates or emergency texts over audio

fparri, 2024-03-21:
A very interesting project: https://github.com/aicodix/rattlegram

kylecorry31, 2024-03-21:
Interesting I'll take a look. I could see this being used for photo map sharing - but it would require the microphone permission which I'm still on the fence about. 

I'll do more research though.

michaelblyons, 2024-03-22:
Does this have practical advantages over QR code sharing? Maybe total max data transfer?

Neat idea, but if you're close enough to transmit data with the phone speaker, I'm not sure taking ten steps closer to share QR codes isn't good enough.

kylecorry31, 2024-03-22:
> Does this have practical advantages over QR code sharing? Maybe total max data transfer?
> 
> Neat idea, but if you're close enough to transmit data with the phone speaker, I'm not sure taking ten steps closer to share QR codes isn't good enough.

Yeah, that's what I was thinking. Maybe it's more convenient and has longer range (I would need to experiment to see what that range is), but QR codes can transmit data faster (that project says 170 B/s, which QR codes can store up to 2953 B in a single code)

fparri, 2024-03-22:
Yes, my idea was more that if you have a portable radio transmitter, you can use it to send messages, coordinates and other info via radio signals to other people around, even if no data is available. I found it neat, but I don't know if the applications could be useful in our context :)

JensKorte, 2024-03-23:
Maybe add Rattlegram to the list of recommended apps. As soon as Rattlegram supports transferring data from/to other apps use Rattlegram as a helper app.

kylecorry31, 2024-03-30:
Going to close this in favor of the Comms plugin in #2285

closed 2024-03-30 by kylecorry31