kylecorry31/Trail-Sense #2288: F-Droid build crash

michaelblyons, 2024-03-22:
v5.9.0 has some kind of build error.

Something in build gradle about File! being deprecated in Java.

kylecorry31, 2024-03-22:
Hm, looks like it is trying to compile with Java 11 but TS uses Java 17. I'll investigate

kylecorry31, 2024-03-22:
Here's the issue on F-Droid to track it: https://gitlab.com/fdroid/fdroiddata/-/issues/3222

kylecorry31, 2024-03-22:
Looks like F-Droid is in the process of upgrading their servers and build failures are expected right now. I don't know what their timeline is but they said to be patient and the build will be retried once their upgrade is complete: https://gitlab.com/fdroid/fdroiddata/-/issues/3222#note_1828212755

kylecorry31, 2024-03-24:
The build has succeeded now, just needs to be indexed by F-Droid to appear in their repo (the normal process). So I think this will be all set. If you don't see the build within a week, let me know (I'll also keep an eye on it).

closed 2024-03-24 by kylecorry31