kylecorry31/Trail-Sense #2298: Ability for users to submit report of sensors built into their device

7eventech77, 2024-03-28:
As title says, users have the ability to submit a report of their device's sensors so others can use this data to see which common devices has sensors such as barometer, gyroscope, and magnetometer built in without having to buy the device and find out that is doesn't have a specific sensor. For example, Samsung Galaxy A13 5G & A14 5G are listed to have a barometer but Galaxy A15 5G not have one listed specifically.  I'm personally looking to purchase a new phone and want to get the 3 main sensors above but not sure which ones have them all or not. Not sure if this could be done offline within trail sense.

kylecorry31, 2024-03-28:
Hi, while I don't think there's a way I can add this directly to Trail Sense, I can support this by doing the following:

- [ ] Add a button to copy all device sensor types (with device model) on the sensor details tool
- [ ] Add a link to a discussion on GitHub where people can post their device details

Also, for reference I have a Galaxy A13 5G and it has all 3 of those sensors and seems to be pretty good quality (a little slow compared to the Galaxy A52 and Pixel 7A, but it seemed to work fine with Trail Sense). I also recently bought a Galaxy A52 5G and it is great (and has all 3 sensors). I can't speak to the A14/A15 though.

kylecorry31, 2024-03-30:
As a start, I've created this page for the devices I've tested on: https://kylecorry.com/Trail-Sense/devices.html

I'll make it easier for people to contribute to that list in the future.

7eventech77, 2024-03-31:
Okay, thank you. With the introduction of the copy/paste of available sensors that should make it much easier to add information to the list.