kylecorry31/Trail-Sense #2327: Use CanvasView for clock

kylecorry31, 2024-04-13:
The clock should be updated to use the CanvasView class.

muryno, 2024-04-14:
@kylecorry31  you mean for both digital and analog?

kylecorry31, 2024-04-14:
No, just the analog clock

muryno, 2024-04-15:
Oooo I get now, instead of using the image as background, should be drawn from the canvas view for analog?

What of digital, since that is a separate view, it can still use the background image right

muryno, 2024-04-15:
In that case, then the background image can be safely deleted, analog already has it background. and we can use the system color for the digital text

muryno, 2024-04-15:
I could pick this up.... and use canvas to also draw digital instead of extra textView...

So ClockView could be reuse for both digital and analog